ovro-lwa/lwa-issues #208: dr-vslow-4 tries to mkdir under /data03?

yupinghuang, 2022-11-29:
```
Nov 28 19:09:00 lwacalim02 dr-vslow-4[597419]: mkdir: cannot create directory ‘/data03’: Permission denied
```

Should be `/data02`?

```
(base) [pipeline@lwacalim02 services]$ systemctl --user status dr-vslow-4.service
● dr-vslow-4.service - OVRO-LWA Slow Visibilties Recorder - Band 4
   Loaded: loaded (/home/pipeline/.config/systemd/user/dr-vslow-4.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2022-11-29 03:09:00 UTC; 30s ago
  Process: 597419 ExecStartPre=/bin/bash -c [ -d /data03/log ] || mkdir -p /data03/log (code=exited, status=1/FAILURE)
 Main PID: 445414 (code=killed, signal=KILL)
```

jaycedowell, 2022-11-29:
Yep, looks like service file has been changed.  If I diff `dr-vslow-4.service` and `dr-vslow-4.service~` I see:
```
< ExecStartPre=/bin/bash -c '[ -d /data03/log ] || mkdir -p /data03/log'
---
> ExecStartPre=/bin/bash -c '[ -d /data02/log ] || mkdir -p /data02/log'
31c31
<          --record-directory       /data03/slow \
---
>          --record-directory       /data02/slow \
34c34
<          --logfile                /data03/log/dr-vslow-4.%H.log'
---
>          --logfile                /data02/log/dr-vslow-4.%H.log'
```

jaycedowell, 2022-11-29:
Original configuration restored and service restarted.

closed 2022-11-29 by jaycedowell