ovro-lwa/lwa-issues #211: datarecorder doesn't write log or send status but still recording data

yupinghuang, 2022-12-02:
After I started data recorders.
```
In [43]: con.status_dr()
Out[43]: 
['12 recording; 4 idle',
 'WARNING: drvs not fully operational: 4 sub-bands have not updated in 120 s: drvs7701,drvs8001,drvs8101,drvs8201',
 'idle',
 'idle',
 'idle',
 'idle',
 'idle',
 'idle',
 'idle',
 'idle']
```
All four data  recorders mentioned are writing measurement sets. But the log files stopped updating since data recording started.
```
-rw-r--r--. 1 pipeline pipeline  4572391 Dec  2 04:07 dr-vslow-3.lwacalim02.log
-rw-r--r--. 1 pipeline pipeline  2199827 Dec  2 04:13 dr-vslow-4.lwacalim02.log
drwxr-xr-x. 2 pipeline pipeline      134 Dec  2 04:13 .
-rw-r--r--. 1 pipeline pipeline  5249148 Dec  2 04:13 dr-beam-3.lwacalim02.log
-rw-r--r--. 1 pipeline pipeline  5233616 Dec  2 04:13 dr-beam-4.lwacalim02.log
-rw-r--r--. 1 pipeline pipeline  5386773 Dec  2 04:13 dr-vfast-3.lwacalim02.log
-rw-r--r--. 1 pipeline pipeline  5385554 Dec  2 04:13 dr-vfast-4.lwacalim02.log
```

Tailing the older log gives
```
(base) [pipeline@lwacalim02 log]$ tail dr-vslow-3.lwacalim02.log
2022-12-02 04:06:34 [DEBUG   ]  max acquire/process/reserve times: 10.022/10.019/0.000
2022-12-02 04:06:34 [DEBUG   ]  receive data rate: 37909762.777 B/s
2022-12-02 04:06:34 [DEBUG   ]  missing data fraction: 0.000%
2022-12-02 04:06:34 [DEBUG   ]  pipeline lag: 30.798465
2022-12-02 04:06:34 [DEBUG   ]  load average: 1.04, 1.16, 1.91
2022-12-02 04:06:34 [DEBUG   ] ===   ===
2022-12-02 04:06:59 [INFO    ] MSStart - Scheduled recording start for 2022-12-02 04:07:14.634523 to 2023-12-02 04:07:14.634523 to /data02/slow/
2022-12-02 04:07:53 [INFO    ] Started operation - <MeasurementSetWriter filename='/data02/slow', start_time='2022-12-02
    04:07:14.634523', stop_time='2023-12-02 04:07:14.634523',
    reduction=None>
```

Status of the service is fine
```
(base) [pipeline@lwacalim02 log]$ systemctl --user status dr-vslow-3
● dr-vslow-3.service - OVRO-LWA Slow Visibilties Recorder - Band 3
   Loaded: loaded (/home/pipeline/.config/systemd/user/dr-vslow-3.service; disabled; vendor preset: enabled)
   Active: active (running) since Fri 2022-12-02 04:04:30 UTC; 16min ago
```

And the datarecorder is writing measurement sets and they seem fine too.

yupinghuang, 2022-12-09:
Here's another instance from dr-vslow-4.lwacalim02.log. Note the time gap after 03:52:53.
```
2022-12-09 03:52:33 [DEBUG   ] === Status Report ===
2022-12-09 03:52:33 [DEBUG   ]  summary: normal
2022-12-09 03:52:33 [DEBUG   ]  info: System operating normally
2022-12-09 03:52:33 [DEBUG   ]  queue size: 1
2022-12-09 03:52:33 [DEBUG   ]  active operation: True
2022-12-09 03:52:33 [DEBUG   ]  active filename: slow
2022-12-09 03:52:33 [DEBUG   ]  active time remaining: 364 days, 23:56:14.198571
2022-12-09 03:52:33 [DEBUG   ] ===   ===
2022-12-09 03:52:43 [DEBUG   ] === Performance Report ===
2022-12-09 03:52:43 [DEBUG   ]  max acquire/process/reserve times: 10.051/10.047/0.000
2022-12-09 03:52:43 [DEBUG   ]  receive data rate: 37815796.714 B/s
2022-12-09 03:52:43 [DEBUG   ]  missing data fraction: 0.000%
2022-12-09 03:52:43 [DEBUG   ]  pipeline lag: 31.833809
2022-12-09 03:52:43 [DEBUG   ]  load average: 2.13, 2.26, 2.88
2022-12-09 03:52:43 [DEBUG   ] ===   ===
2022-12-09 03:52:53 [DEBUG   ] === Performance Report ===
2022-12-09 03:52:53 [DEBUG   ]  max acquire/process/reserve times: 10.005/10.002/0.000
2022-12-09 03:52:53 [DEBUG   ]  receive data rate: 37910827.412 B/s
2022-12-09 03:52:53 [DEBUG   ]  missing data fraction: 0.000%
2022-12-09 03:52:53 [DEBUG   ]  pipeline lag: 31.833809
2022-12-09 03:52:53 [DEBUG   ]  load average: 1.96, 2.21, 2.86
2022-12-09 03:52:53 [DEBUG   ] ===   ===
2022-12-09 04:10:30 [INFO    ] MSStop - Stopped recording started at 2022-12-09 03:48:15.609143 to /data02/slow
2022-12-09 04:10:38 [INFO    ] Ended operation - <MeasurementSetWriter filename='/data02/slow', start_time='2022-12-09
    03:48:15.609143', stop_time='2022-12-09 04:09:50.650444',
    reduction=None>
```

Happened to both the recorders on calim06 as well

```
2022-12-09 03:48:55 [INFO    ] Started operation - <MeasurementSetWriter filename='/data06/slow', start_time='2022-12-09
    03:48:15.664492', stop_time='2023-12-09 03:48:15.664492',
    reduction=None>
2022-12-09 04:10:31 [INFO    ] MSStop - Stopped recording started at 2022-12-09 03:48:15.664492 to /data06/slow
2022-12-09 04:10:39 [INFO    ] Ended operation - <MeasurementSetWriter filename='/data06/slow', start_time='2022-12-09
    03:48:15.664492', stop_time='2022-12-09 04:09:51.003791',
    reduction=None>
```

jaycedowell, 2022-12-09:
Does journalctl show anything on these pipelines?

yupinghuang, 2022-12-09:
Nothing out of the ordinary...

From  `journalctl -S 2022-12-09 -U 2022-12-10`

```
Dec 09 03:47:19 lwacalim06 systemd[7453]: Starting OVRO-LWA Slow Visibilties Recorder - Band 11...
Dec 09 03:47:19 lwacalim06 systemd[7453]: Started OVRO-LWA Slow Visibilties Recorder - Band 11.
Dec 09 03:47:21 lwacalim06 dr-vslow-11[47664]: Read antpos from etcd
Dec 09 03:47:23 lwacalim06 systemd[7453]: Starting OVRO-LWA Slow Visibilties Recorder - Band 12...
Dec 09 03:47:23 lwacalim06 systemd[7453]: Started OVRO-LWA Slow Visibilties Recorder - Band 12.
Dec 09 03:47:24 lwacalim06 dr-vslow-12[47783]: Read antpos from etcd
Dec 09 03:47:24 lwacalim06 dr-vslow-11[47664]: ++++++++++++++++ seq0     = 166539151
Dec 09 03:47:24 lwacalim06 dr-vslow-11[47664]:                  time_tag = 327429295932702720
Dec 09 03:47:24 lwacalim06 dr-vslow-11[47664]: ******** CFREQ: 61250000.0
Dec 09 03:47:24 lwacalim06 dr-vslow-11[47664]: 2022-12-09 03:47:25        INFO        MeasIERS::fillMeas(MeasIERS::Files, Double)         Requested JD 59922.2 is outside the range of the IERS (Earth axis data) >
Dec 09 03:47:24 lwacalim06 dr-vslow-11[47664]: 2022-12-09 03:47:25        INFO        MeasIERS::fillMeas(MeasIERS::Files, Double) +        Calculations will proceed with less precision
Dec 09 03:47:25 lwacalim06 dr-vslow-12[47783]: ++++++++++++++++ seq0     = 166539151
Dec 09 03:47:25 lwacalim06 dr-vslow-12[47783]:                  time_tag = 327429295932702720
Dec 09 03:47:25 lwacalim06 dr-vslow-12[47783]: ******** CFREQ: 65843750.0
Dec 09 03:47:25 lwacalim06 dr-vslow-12[47783]: 2022-12-09 03:47:25        INFO        MeasIERS::fillMeas(MeasIERS::Files, Double)         Requested JD 59922.2 is outside the range of the IERS (Earth axis data) >
Dec 09 03:47:25 lwacalim06 dr-vslow-12[47783]: 2022-12-09 03:47:25        INFO        MeasIERS::fillMeas(MeasIERS::Files, Double) +        Calculations will proceed with less precision
Dec 09 03:47:44 lwacalim06 dr-vslow-12[47783]: /home/pipeline/ovro_data_recorder/dr_visibilities.py:338: RuntimeWarning: divide by zero encountered in log10
Dec 09 03:47:44 lwacalim06 dr-vslow-12[47783]:   im = self._plot_spectra(time_tag, freq, 10*numpy.log10(adata))
Dec 09 03:47:44 lwacalim06 dr-vslow-11[47664]: /home/pipeline/ovro_data_recorder/dr_visibilities.py:338: RuntimeWarning: divide by zero encountered in log10
Dec 09 03:47:44 lwacalim06 dr-vslow-11[47664]:   im = self._plot_spectra(time_tag, freq, 10*numpy.log10(adata))
Dec 09 03:48:56 lwacalim06 dr-vslow-12[47783]: E1209 03:48:56.170321295   47892 fork_posix.cc:76]           Other threads are currently calling into gRPC, skipping fork() handlers
Dec 09 03:48:56 lwacalim06 dr-vslow-12[47783]: E1209 03:48:56.458699416   47892 fork_posix.cc:76]           Other threads are currently calling into gRPC, skipping fork() handlers
Dec 09 03:48:56 lwacalim06 dr-vslow-11[47664]: E1209 03:48:56.888278232   47775 fork_posix.cc:76]           Other threads are currently calling into gRPC, skipping fork() handlers
```

jaycedowell, 2022-12-10:
This might be related to the exception in #213.  The quota management is part of the `GlobalLogger` thread which is responsible for the status/storage/performance log entries.  It would be interesting to know if `dr-vslow-11` and `-12` have a similar exception (or really any exception) in the syslog.

yupinghuang, 2022-12-21:
This happened again just now
```
In [8]: con.status_dr()
Out[8]: 
['recording',
 'WARNING: drvs not fully operational: 6 sub-bands have not updated in 120 s: drvs7801,drvs7802,drvs7901,drvs7902,drvs8301,drvs8302',
 'idle',
 'idle',
 'idle',
 'idle']
```
and no I didn't see anything in syslog or journalctl

yupinghuang, 2022-12-21:
Some time later they seemed to have all recovered
```
In [3]: con.status_dr()
Out[3]: ['idle', 'idle', 'idle', 'idle', 'idle']
```

jaycedowell, 2022-12-21:
This one seems like a slightly different since all the slow recorders are reporting that they are ok and their logs are updating.  What I did notice in the logs is that the "pause" is followed by a quota management report that lots of files were removed.  For example:
```
2022-12-21 03:35:25 [DEBUG   ]  pipeline lag: 32.010443
2022-12-21 03:35:25 [DEBUG   ]  load average: 2.26, 2.65, 3.10
2022-12-21 03:35:25 [DEBUG   ] ===   ===
2022-12-21 03:46:14 [DEBUG   ] === Quota Report ===
2022-12-21 03:46:14 [DEBUG   ] Removed 313 files
2022-12-21 03:46:14 [DEBUG   ] ===   ===
```
or
```
2022-12-21 03:30:01 [INFO    ] Ended operation - <MeasurementSetWriter filename='/data06/slow', start_time='2022-12-21
    03:02:03.637619', stop_time='2022-12-21 03:29:16.518749',
    reduction=None>
2022-12-21 03:43:48 [DEBUG   ] === Quota Report ===
2022-12-21 03:43:48 [DEBUG   ] Removed 62 files
2022-12-21 03:43:48 [DEBUG   ] ===   ===
```
Maybe this is some kind of bad interaction between the writer, the quota manager, and the filesystem?

jaycedowell, 2022-12-22:
I made some changes to the quota management to try and make it less file metadata intensive.  I don't know that that will solve these hangs or if these changes will survive the transition to lustre.  The best solution may be to move the quota stuff out of the data recorders.

caseyjlaw, 2022-12-28:
Seeing perhaps something related today...
Data recorder "drvs7801" and a few others have op-type of `recording`, but no data is being written. All threads in the pipeline seem to be operating correctly and the status is otherwise normal.
I then successfully ran a slow visibility recording run and files were written to disk for this recorder. The state remained in this incorrect state, as before the observation.

jaycedowell, 2023-01-27:
I've disabled quota management for the slow/fast recorders.

yupinghuang, 2023-02-03:
I found
```
In [9]: con.status_dr(['drvs'])
Out[9]: 
['0 recording; 0 idle; 16 timed out',
 'WARNING: drvs not fully operational: 16 sub-bands have not updated in 120 s: drvs7601,drvs7602,drvs7701,drvs7702,drvs7801,drvs7802,drvs7901,drvs7902,drvs8001,drvs8002,drvs8101,drvs8102,drvs8201,drvs8202,drvs8301,drvs8302']

In [10]: con.start_dr(['drvs'])
Starting recorders: ['drvs']
ERROR1: timeout
WARNING: no response from drvs
```
but then the data recorders did record data. The log seems stuck at
```
2023-02-03 23:16:26 [INFO    ] Statistics: Start of new sequence: {'time_tag': 328391370371235840, 'seq0': 167028488, 'chan0': 1216, 'cfreq': 29093750.0, 'nchan': 192, 'bw': 4593750.0, 'navg': 1966080000, 'nstand': 352, 'npol': 4, 'nbl': 62128, 'complex': True, 'nbit': 32}
2023-02-03 23:16:26 [INFO    ] Baseline: Start of new sequence: {'time_tag': 328391370371235840, 'seq0': 167028488, 'chan0': 1216, 'cfreq': 29093750.0, 'nchan': 192, 'bw': 4593750.0, 'navg': 1966080000, 'nstand': 352, 'npol': 4, 'nbl': 62128, 'complex': True, 'nbit': 32}
2023-02-03 23:16:26 [INFO    ] Spectra: Start of new sequence: {'time_tag': 328391370371235840, 'seq0': 167028488, 'chan0': 1216, 'cfreq': 29093750.0, 'nchan': 192, 'bw': 4593750.0, 'navg': 1966080000, 'nstand': 352, 'npol': 4, 'nbl': 62128, 'complex': True, 'nbit': 32}
2023-02-03 23:16:26 [INFO    ] Writer: Start of new sequence: {'time_tag': 328391370371235840, 'seq0': 167028488, 'chan0': 1216, 'cfreq': 29093750.0, 'nchan': 192, 'bw': 4593750.0, 'navg': 1966080000, 'nstand': 352, 'npol': 4, 'nbl': 62128, 'complex': True, 'nbit': 32}
2023-02-03 23:16:46 [INFO    ] Current pipeline lag is 0:00:31.171416
```

jaycedowell, 2023-02-06:
Looks like this instance was related to etcd going down on 2/1.

caseyjlaw, 2023-02-06:
@jaycedowell There was an outage, but also a reset of the etcd keys (massive bloat in number of keys was a concern to Rick). In your testing, is it required that some keys be initialized in order for data recorder m&c to work properly?

jaycedowell, 2023-02-06:
Some pre-existing set of keys really shouldn't be required since everything that is used in determining the overall status is generated by the recorder itself.  This was more of an issue with not properly dealing with missing keys.

caseyjlaw, 2023-02-06:
Is that a data recorder fix or a restart procedure fix? If the latter, please let me know what was expected on your end.

jaycedowell, 2023-02-06:
It's a data recorder fix.  I've already put it in place and restarted the services.

caseyjlaw, 2023-02-06:
Ok, thanks. Closing...

closed 2023-02-06 by caseyjlaw