ovro-lwa/lwa-issues #213: random datarecorder Exception in syslog

yupinghuang, 2022-12-09:
A little confused why it is looking for a data file from October...

```
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]: Exception in thread GlobalLogger:
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]: Traceback (most recent call last):
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]:  File "/home/pipeline/.conda/envs/datarecorder/lib/python3.6/threading.py", line 916, in _bootstrap_inner
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]:    self.run()
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]:  File "/home/pipeline/.conda/envs/datarecorder/lib/python3.6/threading.py", line 864, in run
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]:    self._target(*self._args, **self._kwargs)
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]:  File "/home/pipeline/ovro_data_recorder/monitoring.py", line 552, in main
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]:    self.storage.main(once=True)
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]:  File "/home/pipeline/ovro_data_recorder/monitoring.py", line 250, in main
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]:    self._update()
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]:  File "/home/pipeline/ovro_data_recorder/monitoring.py", line 207, in _update
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]:    self._files.sort(key=lambda x: os.path.getmtime(x))
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]:  File "/home/pipeline/ovro_data_recorder/monitoring.py", line 207, in <lambda>
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]:    self._files.sort(key=lambda x: os.path.getmtime(x))
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]:  File "/home/pipeline/.conda/envs/datarecorder/lib/python3.6/genericpath.py", line 55, in getmtime
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]:    return os.stat(filename).st_mtime
Dec  8 19:52:53 lwacalim02 dr-vslow-4[3162418]: FileNotFoundError: [Errno 2] No such file or directory: '/data02/slow/20221031_164107_29MHz.ms'
```

jaycedowell, 2022-12-10:
It's probably from the quota management feature.

jaycedowell, 2022-12-22:
I'm going to say that this particular issue is solved but there is still the larger issue of quota management in #211.

closed 2022-12-22 by jaycedowell