ovro-lwa/lwa-issues #214: Capture requirements for the transport and accees of local generated files(i.e. png)

rh-codebase, 2022-12-14:
Need a local solution for the transport of generated files, mostly png, jpeg, pdf etc types to allow them to be externally accessed. The SLACK API can possibly be used as a starting point.

caseyjlaw, 2022-12-14:
A good use case is to consider an observer who wants to view plots related to system health or observations. Plots may be generated on any part of the OVRO-LWA network, but we'd like those to be available in a central location that can be exposed via a web server.

rh-codebase, 2022-12-15:
Other requirements TBD: Feel free to add more

- overall storage requirements
- Time to live for each product or kept indefinitely
- allow overwrite of identically named products
- any meta data attached to product
- search requirements. (e.g. find product using meta data)

caseyjlaw, 2023-12-14:
I created a quick image drop viewer through the obs state web app. Does not implement most of these requirements, but people can display images after setting up a tunnel to see the app.

rh-codebase, 2023-12-14:
Nice. Where is the  storage located?

caseyjlaw, 2023-12-14:
Storage currently set to /opt/devel/pipeline/images. App serves that via at lwacalim10:8082/images.