ovro-lwa/lwa-issues #215: PFB flagging

caseyjlaw, 2022-12-14:


jack-h, 2023-01-25:
Implemented in https://github.com/realtimeradio/caltech-lwa/tree/flagging
Ready to roll out when this branch is tested.

caseyjlaw, 2023-02-15:
Proposed change: channel 0 would carry a boolean flag for the whole spectrum. These must be recognized by x-eng, ignored for science, but accumulated.

caseyjlaw, 2023-04-05:
Jayce says that data recorders could use flag count (weight) in data products.

jack-h, 2023-04-05:
I'll summarize what I understand about the second part of this request (counting the number of flags and compensating for it downstream):

1. For each GPU pipeline (there are 32) designate 1 of the pipeline's 96 frequency channels as a "flag" channel.
2. These 32 flag channels are normally set to 1. If the F-engine blanks a spectrum because it exceeds the total power threshold, the whole spectra is set to zero, but, the flag channels are set to 0.
3. Flag channels are not treated in any special way by the correlator
4. Downstream of the correlator (post integration) the integrated values in the flag channels of each visibility contain the number of non-blanked samples which went into this visibility.

A few options exist after this:
1. Rescale the correlations by 1/(fraction of data masked) so that all visibilities look like they had no masked samples
2. Don't do anything to rescale the data, but pass the per-basline integration count downstream to the data recorder, via the [`cor_navg`](https://github.com/realtimeradio/caltech-bifrost-dsp/blob/a9cda5bcca08d2e41d54efa931e078a20134c65e/pipeline/lwa352_pipeline/blocks/corr_output_full_block.py#L264) field of the X-engine -> data recorder packets.

What should be done with the flag channels?
1. Pass them downstream as normal, and let users remember to flag them?
2. Zero them out of the data?
3. Set them to the max allowed value so that they will be flagged by any sane RFI algorithm?
4. Something else?

greghell, 2023-04-05:
I am in favor of passing the flags counters downstream so that we can monitor how much data is lost and for what antennas, but also find a weighting scheme to the visibility to account for that data loss.
Just to be sure, is the "per-basline integration count" is the minimum counter between the two antennas of the baseline?

jaycedowell, 2023-04-05:
I don't know that using `corr_navg` would work.  That field is used internally to determine which sequence a packet belongs to and is then discarded.

jack-h, 2023-04-06:
Is there any field that would work, or is the only easy option to rescale at the correlator and pretend the flagging never happened? (Obviously this doesn't do what @greghell notes above).

@greghell "per-baseline integration count" would be the sum of the correlation of the flag channel. I.e., it increments only when both inputs to the correlation are not masked. I.e., it truly is a count of how much valid data went into each visibility, and properly accounts for whether one, none, or both input signals have been masked (which was the concern LRD raised in the meeting yesterday)

jaycedowell, 2023-04-06:
There aren't a lot of great options here:
 * We could put something in the header but the header only exists long enough to validate the contents of the packet and determine where to put the packet's payload into memory.
 * We could rescale and pretend the flagging never happened but there isn't a way to track this the data recorder side.  Plus, there would probably need to be a change from ci32 to cf32 in the packet payload to deal with the rescaling.
 * We could send N+1 channels in the payload where the first channel is actually the flag counters.  That's messy from a memory handling standpoint but it could work.  You could treat it like new header fields on the send side and as N+1 channels with the old header on the receive side.
 * We could send the flags as a separate packet stream to the data recorders.  At 4 bytes/antpol * 704 antpol that's <3 kB so one packet could contain all flag counters per antenna per integration.  I'm not sure how you deal with two incoming packets streams on the data recorder side.  Maybe you don't and there is a separate piece of software that receives those packets and writes out some kind of flagging or reweighing file.

jaycedowell, 2023-04-06:
Should similar flag counters be implemented for the beam former outputs as well?

jack-h, 2023-04-06:
>     * We could send N+1 channels in the payload where the first channel is actually the flag counters.  That's messy from a memory handling standpoint but it could work.  You could treat it like new header fields on the send side and as N+1 channels with the old header on the receive side.

This is sort of what you'd get if I did nothing, in the sense that the first channel of N would be the flag counter. 

> * We could send the flags as a separate packet stream to the data recorders.  At 4 bytes/antpol * 704 antpol that's <3 kB so one packet could contain all flag counters per antenna per integration. [...]

jack-h, 2023-04-06:
> Should similar flag counters be implemented for the beam former outputs as well?

Maybe? I guess the logic being applied is that this is most useful for the slow correlator (where post-processing can only flag whole 10s chunks). But, given the flagging is happening on the F-engines, all the modes will be subject to the flagged data, and will have the flag channels appearing in their data streams whether they want that or not

jack-h, 2023-12-01:
Closing this -- follow up discussion in #389

closed 2023-12-01 by jack-h