ovro-lwa/lwa-issues #217: Are the 40GbE switches ok?

yupinghuang, 2022-12-16:
From lwacalim02 (10.41.0.77), I can't ping 00 or 09 but other nodes are fine. I knew that connecting to calim00 on the 40GbE interface from other calim nodes used to work since I've been using that...

```
(base) [yuping@lwacalim02 ~]$ ping 10.41.0.84
PING 10.41.0.84 (10.41.0.84) 56(84) bytes of data.
From 10.41.0.77 icmp_seq=1 Destination Host Unreachable
```
```
(base) [yuping@lwacalim02 ~]$ ping 10.41.0.85
PING 10.41.0.85 (10.41.0.85) 56(84) bytes of data.
64 bytes from 10.41.0.85: icmp_seq=1 ttl=64 time=0.991 ms
```
```
(base) [yuping@lwacalim02 ~]$ ping 10.41.0.75
PING 10.41.0.75 (10.41.0.75) 56(84) bytes of data.
From 10.41.0.77 icmp_seq=1 Destination Host Unreachable
```
None one can ping 10.41.0.75 btw.

calim00 (and 09) still thinks that it has the 40GbE card and the IP address so it's probably not a 
```
(datarecorder) [pipeline@lwacalim00 fast]$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp68s0f0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 3c:ec:ef:61:3a:2e brd ff:ff:ff:ff:ff:ff
    inet 10.40.0.20/24 brd 10.40.0.255 scope global noprefixroute enp68s0f0
       valid_lft forever preferred_lft forever
    inet6 fe80::3eec:efff:fe61:3a2e/64 scope link 
       valid_lft forever preferred_lft forever
3: enp68s0f1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 3c:ec:ef:61:3a:2f brd ff:ff:ff:ff:ff:ff
4: enp161s0f0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 9000 qdisc mq state DOWN group default qlen 1000
    link/ether 40:a6:b7:5a:6c:38 brd ff:ff:ff:ff:ff:ff
    inet 10.41.0.75/26 brd 10.41.0.127 scope global noprefixroute enp161s0f0
       valid_lft forever preferred_lft forever
    inet6 fe80::42a6:b7ff:fe5a:6c38/64 scope link 
       valid_lft forever preferred_lft forever
5: enp161s0f1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 40:a6:b7:5a:6c:39 brd ff:ff:ff:ff:ff:ff
```

rh-codebase, 2022-12-16:
the 40G subnet on the calim and gpu nodes is defined as 10.41.0.0/26 inside a 10.41.0.0/24. All bets are off regarding ping and sane networking. This config works for what Jack needs it for. It's possible that at some past time, the calim nodes interface was on 10.41.0.0/24 which would explain why expected networking was seen.

yupinghuang, 2022-12-16:
That's...unfortunate. The calim nodes will need to talk to the storage nodes via their 40 gbe interface.

All other calim nodes were able to talk to 10.41.0.75 (calim00) until 12/7. I'm confused why the GPU nodes can talk to, say 10.41.0.76 (calim01), from, say, 10.41.0.72 (gpu08), but no one can talk to 10.41.0.75? Maybe I just don't understand this whole thing...shouldn't all the calim 40 gbe sit in the 10.41.0.65/26 subnet?

rh-codebase, 2022-12-19:
I see what you are saying now. 10.41.0.75 should behave like any other 10.41.0.0/26

yupinghuang, 2022-12-21:
Yeah it still does not. Still having the same issues in the OP

rh-codebase, 2022-12-22:
link is not coming up. same on other switch ports so likely the driver or card. the host has not been rebooted since the opening of this issue. can it be rebooted? if so, when?

yupinghuang, 2022-12-22:
I'll reboot it this afternoon and report back unless you'd like to do it yourself.

rh-codebase, 2022-12-22:
I can do it now  if that works for you

yupinghuang, 2022-12-22:
now's a good time

rh-codebase, 2022-12-22:
if reboot doesn't fix, I'll reseat the card.

yupinghuang, 2022-12-22:
btw calim09 is like this too.

rh-codebase, 2022-12-22:
intel 40G SFP module from fs.com  failed on calim00. replaced. interface up
switchport misconfigured for calim09. moved to switchport 15 and config for 40g. interface up.

closed 2022-12-22 by rh-codebase