ovro-lwa/lwa-issues #218: Solar beamforming issue

dgary50, 2022-12-16:
I attempted to take some solar data using mnc--not sure if that is supposed to be operational or not.  When I tried to configure the beamformer with the target name 'Sun' (also tried 'sun') I got this:

con.control_bf(num=1,targetname='Sun',track=True)

~/proj/lwa-shell/mnc_python/mnc/control.py in control_bf(self, num, coord, coordtype, targetname, track)
    291
    292         if targetname is not None:
--> 293             self.bfc[num].set_beam_target(targetname)
    294         elif ra is not None:
    295             self.bfc[num].set_beam_target(ra, dec=dec)

KeyError: 1

I then tried using HA and Dec with con.control_bf(num=1,targetname=(30.0, -22.5),track=True) but got the same error.

Seeing the "elif ra is not None" line, I also tried con.control_bf(num=1,ra=264.0, dec=-23.4,track=True) but got

----> 1 con.control_bf(num=1,ra=264.0, dec=-23.4,track=True)
TypeError: control_bf() got an unexpected keyword argument 'ra'

caseyjlaw, 2022-12-16:
Thanks, Dale. The `mnc-python` code should work for this purpose.
Were you using the system use recipe? It describes a few setup steps, as well as debugging/status steps to help understand what's happening. My guess is that you skipped the `configure_xengine` step.

dgary50, 2022-12-16:
No, I followed all of the steps in the use recipe, I think.  I let it use
the standard config file and it seemed to report success on that.

--Dale

closed 2023-10-19 by caseyjlaw