ovro-lwa/lwa-issues #219: journalctl not working on cal-im

caseyjlaw, 2022-12-19:
I think the logging on cal-im isn't set up quite right. I'd like to look a logs at some point in the past (to resolve #202). However, systemctl only shows the latest log statement and journalctl has permissions issue:

```
journalctl --user-unit dr-vslow-1
Hint: You are currently not seeing messages from the system.
      Users in the 'systemd-journal' group can see all messages. Pass -q to
      turn off this notice.
No journal files were opened due to insufficient permissions.
```

yupinghuang, 2022-12-19:
assign to myself. You can run with sudo and just look for the unit instead of user unit. It is not clear to me that there's a way to make it work with user units but I'll try adding pipeline to the systemd-journal group.

yupinghuang, 2023-02-24:
Everybody now added to systemd-journal group. I also updated the new user creation script so that new user gets added.

closed 2023-02-24 by yupinghuang