ovro-lwa/lwa-issues #222: Beamforming without duration definition

iveydavis, 2022-12-21:
It'd be nice to do beamforming via mnc control without having to define the duration of the observation. Right now, stopping the observation prematurely (at least by stopping the data recording service) prevents the hdf5 file from being written properly.

jaycedowell, 2022-12-21:
The power beam recorders have a 'cancel' command which can be used to stop a recording.  This isn't currently exposed through `mnc.control`.

caseyjlaw, 2022-12-27:
I could implement this. Two issues come up:
- We could default beamforming to record "endlessly" (e.g., 24 hrs as default), then the data will be written as one large file. As I recall, that makes disk management tricky. Ivey, would you be ok with one large file or do you want a change here?
- Jayce, I see that the cancel command uses a "queue number" to refer to an ongoing recording. Is that number available somewhere? Can one assume it starts at 0 and increments only when there are multiple submitted to the queue?

jaycedowell, 2022-12-27:
Yes, the queue number starts at zero and increments when there are multiple recordings scheduled.  In the case of multiple records zero will always be the first in time.  I thought I had a way to look up what the queue entries are but I cannot find it in the list of monitoring points.

Alternatively, the cancel command supports canceling by filename (technically the basename of the file).  The active recording file can be queries through the "op-tag" monitoring point.

caseyjlaw, 2022-12-27:
Would this syntax be correct to cancel (and close file writing) of the power beam 1?
`accepted, response = client.send_command('dr1', 'cancel', mjd='now', mpm='now', queue=0)`

jaycedowell, 2022-12-27:
`accepted, response = client.send_command('dr1', 'cancel', queue_number=0)`

caseyjlaw, 2022-12-28:
Added the 'cancel' syntax to the `Control.stop_dr` method. Confirmed with a test.

closed 2022-12-28 by caseyjlaw