ovro-lwa/lwa-issues #223: add more settings control to lwa-cfg

caseyjlaw, 2022-12-28:
The ARX and F-engine subsystems have optimal settings that are saved in matlab binaries. These control attenuator settings and f-engine digital gain. We need to integrate these settings with the lwa-cfg system (https://github.com/ovro-lwa/lwa-cfg).

Data are at https://github.com/ovro-lwa/mnc_python/tree/main/data. ARX setting script is at https://github.com/ovro-lwa/mnc_python/blob/main/scripts/arxSettingsLoad.py. F-eng eq setting script is at https://github.com/ovro-lwa/mnc_python/blob/main/scripts/fenginesLoadByDsig.py. Both scripts take the matlab binary as an argument.

It is not clear to me whether it is worth the effort to convert these files into yml, since settings and script already work as is. I would be tempted to have the /cfg callback simply execute the existing script and settings file. We could then track changes by using unique settings file names.
The alternative is to rewrite all the setting code to parse from etcd, which seems fragile to me. Open to discussion.

caseyjlaw, 2022-12-28:
This may also help read matlab files: https://github.com/ewiger/yamlmatlab.

yupinghuang, 2023-03-07:
Comments welcomed. Maybe it's time to revisit  this.

caseyjlaw, 2023-03-08:
`mnc-python` has [a script](https://github.com/ovro-lwa/mnc_python/blob/main/scripts/loadFengineAndArxSettings.py) that sets key ARX and f-eng settings. This supersedes the two scripts described above.
Two open issues to get that integrated with this system:
1. Deciding on what gets set to /cfg
2. Not clear to me that any services are watching /cfg.

caseyjlaw, 2023-04-12:
Proposal for rough requirements on configuration:
1. ARX, F-engine, and X-engine configuration settings can be set programmatically by observers or from an automated process.
2. Changes to the settings will be tracked, so that users can find settings for data recorded in the past.
3. Key scenario includes setting and tracking changes to EQ/attenuator settings at day/night changes.
4. Changes will be tracked on a cadence of ~10 seconds.
5. Subsystems will reload parameters upon changes.

caseyjlaw, 2023-04-14:
Before I forget, the 
x-engine configuration (if in /cfg) could also include new configuration. For example, we once got bitten by the accumulation length. This could be set as `autocorr.set_acc_len=2^15`.