ovro-lwa/lwa-issues #225: The times being recorded for data (including filenames) are off by ~100 minutes

gregghallinan, 2023-01-06:
The UTC times recorded to the metadata in the MS files are currently >10o minutes later than the UTC times at which those data are collected. The same applies to the filenames. This looks o be an issue that began last night and has increasingly got worse.

jaycedowell, 2023-02-01:
What day was this?

For what it is worth I didn't run into this when I was testing the data recorders last week.  That was also starting from a state where the x-engines and data recorders had just been restarted.

yupinghuang, 2023-02-03:
1/5. It was after the correlator was kep running for a while (~a week?). A correlator restart seemed to have fixed it

jaycedowell, 2023-08-31:
@gregghallinan do you have a recent example of this I can look at?

yupinghuang, 2023-09-02:
It's probably related to https://github.com/ovro-lwa/lwa-issues/issues/465.

jack-h, 2023-09-06:
@yupinghuang why do you think issue #465  is related to this one? (it might be, but #465  seems to just kill the slow corr output all together)

yupinghuang, 2023-09-27:
@jack-h part of it is the timing of Andrea reporting wrong timestamps in https://github.com/ovro-lwa/lwa-issues/issues/464 and #465. My theory is that the correlator started a new sequence but the data recorder didn't register that.

ok this does not account for the "has increasingly got worse" bit but 1/5 was a long while ago.