ovro-lwa/lwa-issues #230: Robust, simple initialization procedure

caseyjlaw, 2023-02-08:


caseyjlaw, 2023-02-08:
`mnc-python` now has script `restoreSettings.sh` that will set some basic settings.

yupinghuang, 2023-02-22:
I'm hoping that with https://github.com/lwa-project/ovro_data_recorder/pull/23 we won't ever have to restart data recorder service again. Since we'll only have one mode for the 24-hour run this seems like a no-op, except that we need

- the go/no go thing to tell us that we're ok, and
- maybe don't restart the correlator all the time if it's already on (check that it's on and it's configured to send data the way we want. Long term we should get rid of `configure_xengine` since it just sends command for places to send data, which we should be able to check when setting up the DR).
- calculating delays for tracking really shouldn't block the ipython session. Or just swap the data recorder start and control_bf steps in the system use recipe when beamforming.
-

caseyjlaw, 2023-03-01:
Go/no-go in another issue.
Delays and beamforming item can be handled in docs/procedures.

closed 2023-03-01 by caseyjlaw