ovro-lwa/lwa-issues #231: Display beam pointing and frequencies

caseyjlaw, 2023-02-08:


caseyjlaw, 2023-02-08:
Nice past work on all-sky image for planning: https://github.com/ovro-lwa/low-freq-sky

jaycedowell, 2023-02-27:
The [ovro_data recoder `diagnostic_image` branch](https://github.com/lwa-project/ovro_data_recorder/tree/diagnostic_image) adds a new `ImageOp` to the slow visibilities recorders that generates an image using the core and 100 kHz of bandwidth every minute.  These images can be accessed through the "diagnostics/image" monitoring point.

yupinghuang, 2023-02-27:
Can we add a monitoring point for alt/az for each beam (https://github.com/ovro-lwa/mnc_python/blob/main/mnc/xengine_beamformer_control.py#L344, say, push here) and then, say, put circles on the image indicating where the active beams are?

caseyjlaw, 2023-02-27:
That's a good idea. I looked at the code and the delays are currently loaded directly (no RA, Dec is passed through etcd), so it would be a new monitor point.
We might want to think more broadly about the design here. For example, do we want any scheduled observations tracked? That might motivate monitor points named differently, e.g., `/cmd/beam/n` versus `/cmd/observation/`.