ovro-lwa/lwa-issues #232: Display observing problems on web page for observers

caseyjlaw, 2023-02-08:
A dashboard could highlight observing problems relevant to observing and analysis (e.g., antenna flags).

caseyjlaw, 2023-02-15:
Related: low-freq-sky displays what sources are up and informs analysis. @yupinghuang

caseyjlaw, 2023-03-15:
Go/no-go and real-time narrow-band image are working.

We need a display of:
- delay solutions
- bad antennas

caseyjlaw, 2023-04-04:
Parsing caltable (bcal) files to get list of flagged antennas. added table pane showing these flags.
Found syntax to display delays per ant-pol.

yupinghuang, 2023-04-10:
https://github.com/ovro-lwa/lwa-issues/issues/245

this should probably run as a service, but here's where it's at.

caseyjlaw, 2023-12-14:
We have many new services to expose problems. Big ones are logger, system health dashboard, and obs state web app.

closed 2023-12-14 by caseyjlaw