ovro-lwa/lwa-issues #234: Power level monitoring in grafana

caseyjlaw, 2023-02-08:
ADC saturation and FFT overflow should be ingested to influx/grafana.
This will provide understanding of data quality for current observer or for past observations.

caseyjlaw, 2023-03-15:
F-eng data going into influx and grafana dashboard.
Need code to pull it out for flagging individual integrations.

rh-codebase, 2023-03-30:
influx measurements  for power level monitoring. 1 minute cadence into 'snapmon3' and 4sec cadence into fenghires. Each have host and chan as tags.

snapmon3 contains:

**time, acc_len, chan, clip_count,  fft_shift, flag_threshold, host,   mjd, overflow_count, rms**

fenghires contains:

**time,  chan, host,   means,      mjd,               powers,     rmss,**

caseyjlaw, 2023-07-27:
Moving this to #437

closed 2023-07-27 by caseyjlaw