ovro-lwa/lwa-issues #242: Go/no-go monitor point

caseyjlaw, 2023-02-08:
Need to create a "go/no-go" status. This is a one-stop, high-level key for subsystems, plus their logical AND (maybe ‘/mon/<subsystem>/status’)

jack-h, 2023-02-20:
So, for F-engines, how's about

jack-h, 2023-02-20:
OK -- new etcd key `/mon/snap/<id>/status` created which is a dictionary containing fields `is_ok` and `timestamp`. (Easy to change to standardize with other systems). `is_ok` is True if no blocks in the system return `ERROR` for any of their `get_status()` fields. These fields now include items 1-3 in the F-engine section above.

Access as:

jack-h, 2023-02-20:
And X-engines should now be properly reporting gbps:

```
In [61]: from lwa352_pipeline_control import Lwa352PipelineControl

In [62]: p = Lwa352PipelineControl(host='lxdlwagpu03', pipeline_id=0)

In [63]: p.capture.get_bifrost_status()['gbps']
Out[63]: 12.960346784138247
```

yupinghuang, 2023-02-20:
> 

tbh I'd be fine with just identifying and dashboarding key performance metrics and then the observer can manually do their logical AND (say from the dashboard; or the dashboard can just do it). 

I know this is not what Casey wrote in the issue. But maybe we'd run into a situation where a sub system may be ok for 5 min and then bad for 5 min. It'd be clear from grafana (when we have a dashboard) that this is the case, but a "status ok" can be misleading.

jack-h, 2023-02-20:
Yeah, that seems fair. I think at the very least the three listed F params:

caseyjlaw, 2023-02-20:
This is a great start, Jack. I agree that there are subtleties, but starting with the biggest class of failure modes helps. We can refine as we learn more.

caseyjlaw, 2023-03-04:
I took a first pass at this on the dashboard. Need to add labels at least. Comments welcome.

closed 2023-03-08 by caseyjlaw

caseyjlaw, 2023-03-27:
@jack-h I'm getting a `ok=False` for most f-engines. Looking at the fields you mentioned above, I cannot find 'gbps' under the `eth` block. Was there a change there or is there another reason it is not passing?

jack-h, 2023-03-27:
I'm confused: I see the OK false messages, but I get gbps reporting fine.

```
In [10]: for i in range(1,12):
    ...:     f = snap2_fengine.Snap2FengineEtcd('snap%.2d' % i)
    ...:     print(i)
    ...:     print(f.eth.get_status())

2023-03-27 20:13:55,813 - lwa_f.blocks.block:snap01 - WARNING - eth - Couldn't find Ethernet core. Will retry later
1
[{'tx_of': 0, 'tx_full': 16, 'tx_vld': 3847160015, 'tx_ctr': 2445589483, 'gbps': 37.8693461901592}, {'tx_full': 2}]
2023-03-27 20:13:55,942 - lwa_f.blocks.block:snap02 - WARNING - eth - Couldn't find Ethernet core. Will retry later
2
[{'tx_of': 0, 'tx_full': 16, 'tx_vld': 2987054243, 'tx_ctr': 2374371958, 'gbps': 37.80131565118712}, {'tx_full': 2}]
2023-03-27 20:13:56,070 - lwa_f.blocks.block:snap03 - WARNING - eth - Couldn't find Ethernet core. Will retry later
3
[{'tx_of': 0, 'tx_full': 17, 'tx_vld': 1722121543, 'tx_ctr': 2301056589, 'gbps': 37.754754371789105}, {'tx_full': 2}]
2023-03-27 20:13:56,202 - lwa_f.blocks.block:snap04 - WARNING - eth - Couldn't find Ethernet core. Will retry later
4
[{'tx_of': 0, 'tx_full': 16, 'tx_vld': 1682024549, 'tx_ctr': 2234087641, 'gbps': 37.817176980298456}, {'tx_full': 2}]
2023-03-27 20:13:56,331 - lwa_f.blocks.block:snap05 - WARNING - eth - Couldn't find Ethernet core. Will retry later
5
[{'tx_of': 0, 'tx_full': 17, 'tx_vld': 1630558620, 'tx_ctr': 2167059964, 'gbps': 37.814870799399465}, {'tx_full': 2}]
2023-03-27 20:13:56,481 - lwa_f.blocks.block:snap06 - WARNING - eth - Couldn't find Ethernet core. Will retry later
6
[{'tx_of': 0, 'tx_full': 17, 'tx_vld': 770632427, 'tx_ctr': 2095843005, 'gbps': 37.78956085609536}, {'tx_full': 2}]
2023-03-27 20:13:56,611 - lwa_f.blocks.block:snap07 - WARNING - eth - Couldn't find Ethernet core. Will retry later
7
[{'tx_of': 0, 'tx_full': 17, 'tx_vld': 3405259725, 'tx_ctr': 2020479369, 'gbps': 37.81817834867723}, {'tx_full': 2}]
2023-03-27 20:13:56,740 - lwa_f.blocks.block:snap08 - WARNING - eth - Couldn't find Ethernet core. Will retry later
8
[{'tx_of': 0, 'tx_full': 17, 'tx_vld': 1330680854, 'tx_ctr': 1942969020, 'gbps': 37.8162979156983}, {'tx_full': 2}]
2023-03-27 20:13:56,869 - lwa_f.blocks.block:snap09 - WARNING - eth - Couldn't find Ethernet core. Will retry later
9
[{'tx_of': 0, 'tx_full': 16, 'tx_vld': 66504494, 'tx_ctr': 1869657628, 'gbps': 37.799002314259255}, {'tx_full': 2}]
2023-03-27 20:13:56,998 - lwa_f.blocks.block:snap10 - WARNING - eth - Couldn't find Ethernet core. Will retry later
10
[{'tx_of': 0, 'tx_full': 16, 'tx_vld': 3753016221, 'tx_ctr': 1777490155, 'gbps': 37.75667566755264}, {'tx_full': 2}]
2023-03-27 20:13:57,128 - lwa_f.blocks.block:snap11 - WARNING - eth - Couldn't find Ethernet core. Will retry later
11
[{'tx_of': 0, 'tx_full': 17, 'tx_vld': 3702086854, 'tx_ctr': 1710465159, 'gbps': 37.79790685501702}, {'tx_full': 2}]
```

jack-h, 2023-03-27:
If you are using the etcd interface I don't think this should happen. If you aren't, updating the control library will probably fix this.

I'm seeing `fw_supported = False`: the running firmware is 7.3, but `control_sw/src/__fwversion__.py` contains `7.3`. I'm guessing this is @kathrynplant 's update? This file needs updating (assuming there is high confidence the software will continue to work!) and then the etcd daemons need restarting

```
In [20]: f.fpga.get_status()
Out[20]: 
[{'programmed': True,
  'flash_firmware': 'snap2_f_200msps_64i_4096c_2023-03-23_1851.fpg',
  'flash_firmware_md5': 'b963fe78e8da5cfe52869016d95ce107',
  'timestamp': '2023-03-27T20:18:23.452585',
  'host': 'snap02',
  'sw_version': '1.0.1-v1.0.0-150-g80590c5b',
  'fpga_clk_mhz': 195.6417139362378,
  'fw_version': '7.3.0.0',
  'fw_build_time': '2023-03-23T18:51:17',
  'fw_supported': False,
  'temp': 35.764250781249984,
  'vccint': 0.90234375,
  'vccaux': 1.7548828125,
  'vccbram': 0.90234375,
  'sys_mon': 'reporting'},
```