ovro-lwa/lwa-issues #244: Data capture of specific subbands is written to the wrong node

rlbyrne, 2023-02-08:
I began data capture of slow visibilities for two subbands (70 MHz and 75 MHz) following the System Use Recipe. I used the following commands:
```
from mnc import control
con = control.Controller('config/lwa_config_calim.yaml', xhosts=['lxdlwagpu07'])
con.stop_xengine()
con.configure_xengine(recorders=['drvs'])
con.start_dr(recorders=['drvs'])
```

Typically, these subbands are written to `lwacalim07`, but in this case they were written to `lwacalim01`. Moving forward, the data should be written to consistent locations regardless of how many subbands are captured at once.

caseyjlaw, 2023-02-09:
Version 0.6.0 of `mnc-python` should allow arbitrary selection of x-engine hosts that map to the same numbered cal-im node. The configuration file also allows for some remapping of gpu->calim.
Only rudimentary testing, so I'll leave this issue open for now.

jaycedowell, 2023-02-11:
An alternative would have been to configure the system as usual with all gpu servers and then send recording commands only to the data recorders recording those subbands.

caseyjlaw, 2023-02-14:
I noticed that `mnc-python` did not expose this. I made a change and reinstalled in the datarecorder environment.

closed 2023-02-14 by caseyjlaw