ovro-lwa/lwa-issues #245: Script for beamformer users to run to automatically generate new calibration tables

marinanderson, 2023-02-08:
I'm envisioning a script that would do the following:
- run the observing script to collect a few minutes of slow-vis data
- run a simple (Cas/Cyg) calibration pipeline on a single (multiple?) integration(s) for every subband.
- copy the calibration tables to the location where the beamformer looks for the most recent calibration tables
- produce images for each subband to give the user a sense of how well the calibration performed for each subband.

The user would have to know to run this script during a specific LST window in order to get good-enough solutions in advance of their beamformer observations. ANOTHER IDEA FOR THE MONITOR DASHBOARD: plot of Cas/Cyg elevation as a function of UTC / LST / PT??? Or revive the old lwa-sky tool created by S. Bourke.

marinanderson, 2023-02-08:
Another to-do: 
- how frequently should these be updated? in advance of every observation? once per day?
- evidence from a prior observation that solutions remained valid after FPGA reprogram/initialization...would be good to verify this

yupinghuang, 2023-02-08:
I was picturing getting some data every time Cyg A transit (or something having to do with the Sun etc etc), and then the calib pipeline runs, and then it QAs and send the metriccs it's supposed to send.

The script can just run as a service and figure out when it needs to wake up next before it goes to sleep.

Probably we only need this once every few days for beamforming. But it's probably good to run it everyday anyway.

I can work on this and try to get it done by next Thu. I got orca working on calim. Assign this to me if it sounds like a plan..

yupinghuang, 2023-03-15:
As pipeline, something like...

```
pdsh -w lwacalim[01-08] 'conda activate py38_orca && cd /opt/devel/pipeline/distributed-pipeline/ && pipenv run python orca/extra/tempcal.py 20230305_184136 183'
```

where the first argument is the timestamp for the calibration intgration, second argument is the ref ant (Gregg likes 202?), an optional `--image` argument makes image. It prints scatter for delays and puts the bcal sols in `~/caltables/staging`.

It uses `/home/pipeline/bad_ants_latest.txt` for antenna flags...

ref https://github.com/ovro-lwa/distributed-pipeline/blob/main/orca/extra/tempcal.py