ovro-lwa/lwa-issues #246: dr-vslow: missing data fraction: -100.000%

yupinghuang, 2023-02-08:
```
2023-02-08 21:43:50 [DEBUG   ] === Performance Report ===
2023-02-08 21:43:50 [DEBUG   ]  max acquire/process/reserve times: 10.043/10.017/0.021
2023-02-08 21:43:50 [DEBUG   ]  receive data rate: 75868113.773 B/s
2023-02-08 21:43:50 [DEBUG   ]  missing data fraction: -100.000%
2023-02-08 21:43:50 [DEBUG   ]  pipeline lag: 11.825325
2023-02-08 21:43:50 [DEBUG   ]  load average: 2.15, 2.59, 2.84
```
on calim01. like_bmon.py shows 100%.

jaycedowell, 2023-02-09:
It goes well with like_bmon.py showing that 18 EB are missing.

jaycedowell, 2023-02-09:
From the logs there was a restart/change in the data stream at 2/8 1:02:13:
```
2023-02-08 01:02:13 [INFO    ] Statistics: Start of new sequence: {'time_tag': 328460352037847040, 'seq0': 167063572, 'chan0': 2944, 'cfreq': 70437500.0, 'nchan': 192, 'bw': 4593750.0, 'navg': 1966080000, 'nstand': 352, 'npol': 4, 'nbl': 62128, 'complex': True, 'nbit': 32}
2023-02-08 01:02:13 [INFO    ] Baseline: Start of new sequence: {'time_tag': 328460352037847040, 'seq0': 167063572, 'chan0': 2944, 'cfreq': 70437500.0, 'nchan': 192, 'bw': 4593750.0, 'navg': 1966080000, 'nstand': 352, 'npol': 4, 'nbl': 62128, 'complex': True, 'nbit': 32}
...
2023-02-08 01:02:13 [INFO    ] Current pipeline lag is 0:00:11.309543
```

~40 seconds later the missing fraction jumps to 100% and `dr-vslow-1` goes into error:
```
2023-02-08 01:02:52 [DEBUG   ]  missing data fraction: 100.000%
...
2023-02-08 01:02:52 [DEBUG   ]  summary: error
2023-02-08 01:02:52 [DEBUG   ]  info: Packet loss during receive >10% (100.0% missing)
```
and then ~10 s after that the missing fraction goes negative.  This is somehow ok with `dr-vslow-1` and the error condition gets cleared:
```
2023-02-08 01:03:02 [DEBUG   ]  missing data fraction: -100.000%
...
2023-02-08 01:03:13 [DEBUG   ]  summary: normal
2023-02-08 01:03:13 [DEBUG   ]  info: Error condition(s) cleared
```

Something else strange is that before the restart/change the data rate reported in the logs is what I would expect but it roughly doubles when the missing fraction goes negative.  `dr-vslow-2` shows a similar behavior.

There was another restart/change to the data at 2/9 1:18:56 and everything looks ok after that point.

caseyjlaw, 2023-02-09:
This was for a 24-hr run with a single x-engine configured. That should have sent two subbands from gpu07 to lwacalim07, but there is a bug in `mnc-python` that sent the data to lwacalim01. Perhaps that's related to the bad accounting on the data recorders?

jaycedowell, 2023-02-09:
Maybe?  That might happen if both gpu01 and gpu07 were sending slow packets to `dr-vslow-1` and `dr-vslow-2`.

yupinghuang, 2023-02-09:
This used to work when gpu02 was taken down for reprovisioning. I mean it's sending the data to the wrong nodes but there were data. The timing of the errors do look a lil sus tho.

caseyjlaw, 2023-02-09:
I updated `mnc-python` to version 0.6.0 and it should map the data properly now. We can test this again soon.

yupinghuang, 2023-02-10:
Continuing the SAGA. Ruby did a full band data recording test and the timestamps on lwacalim07 are offset from other nodes by 1s...
```
(base) [pipeline@lwacalim01 ~]$ pdsh -w lwacalim[01-08] 'ls -lstra /data0?/slow/ | grep ms |tail -2'
lwacalim07:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:43 20230210_184244_75MHz.ms
lwacalim07:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:43 20230210_184244_70MHz.ms
lwacalim08:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:43 20230210_184245_84MHz.ms
lwacalim08:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:43 20230210_184245_80MHz.ms
lwacalim04:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:43 20230210_184235_47MHz.ms
lwacalim04:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:43 20230210_184235_43MHz.ms
lwacalim05:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:43 20230210_184245_52MHz.ms
lwacalim05:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:43 20230210_184245_57MHz.ms
lwacalim01:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:42 20230210_184205_15MHz.ms
lwacalim01:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:42 20230210_184205_20MHz.ms
lwacalim03:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:43 20230210_184245_38MHz.ms
lwacalim03:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:43 20230210_184245_34MHz.ms
lwacalim02:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:43 20230210_184245_29MHz.ms
lwacalim02:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:43 20230210_184245_24MHz.ms
lwacalim06:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:43 20230210_184245_66MHz.ms
lwacalim06:    9 drwxr-xr-x.    15 pipeline pipeline    20 Feb 10 18:43 20230210_184245_61MHz.ms
```

jaycedowell, 2023-02-10:
Is that also true for the timestamps inside the measurement sets?

yupinghuang, 2023-02-10:
Looks like it's true
```
CASA <1>: listobs('20230210_184154_70MHz.ms')
Out[1]: 
{'BeginTime': 59985.77904532607,
 'EndTime': 59985.77916142585,
 'IntegrationTime': 10.031021118164062,
 'field_0': {'code': 'none',
  'direction': {'m0': {'unit': 'rad', 'value': -1.0019236445338702},
   'm1': {'unit': 'rad', 'value': 0.6455918428881909},
   'refer': 'J2000',
   'type': 'direction'},
  'name': 'zenith'},
 'nfields': 1,
 'numrecords': 62128,
 'scan_1': {'0': {'BeginTime': 59985.77904532607,
   'EndTime': 59985.77916142585,
   'FieldId': 0,
   'FieldName': 'zenith',
   'IntegrationTime': 10.03102040816319,
   'SpwIds': array([0]),
   'StateId': -1,
   'nRow': 62128,
   'scanId': 1}},
 'timeref': 'UTC'}

CASA <2>: listobs('20230210_184155_15MHz.ms')
Out[2]: 
{'BeginTime': 59985.77905531065,
 'EndTime': 59985.77917141043,
 'IntegrationTime': 10.031021118164062,
 'field_0': {'code': 'none',
  'direction': {'m0': {'unit': 'rad', 'value': -1.0018607915414997},
   'm1': {'unit': 'rad', 'value': 0.645591720293518},
   'refer': 'J2000',
   'type': 'direction'},
  'name': 'zenith'},
 'nfields': 1,
 'numrecords': 62128,
 'scan_1': {'0': {'BeginTime': 59985.77905531065,
   'EndTime': 59985.77917141043,
   'FieldId': 0,
   'FieldName': 'zenith',
   'IntegrationTime': 10.03102040816319,
   'SpwIds': array([0]),
   'StateId': -1,
   'nRow': 62128,
   'scanId': 1}},
 'timeref': 'UTC'}
```

jaycedowell, 2023-02-10:
Yep, hrm.

If I ask the x-engines what they are up in terms of the correlator accumulator with:
```
from lwa352_pipeline_control import Lwa352PipelineControl
p=Lwa352PipelineControl('lxdlwagpu01')
q=Lwa352PipelineControl('lxdlwagpu07')
print(p.corr_acc.get_bifrost_status()['stats']['last_end_sample'])
print(q.corr_acc.get_bifrost_status()['stats']['last_end_sample'])
```
I get that  gpu01 is at 40101319776000 and gpu07 is at 40101319755360.  That's a difference of 20640 F-engine windows with gpu07 lagging.  Converting that to time that's about 0.863 s.  If I difference the `BeginTime`s in those two measurement sets I also get 0.863 s.

jaycedowell, 2023-02-10:
The reported correlator accumulator start times are also off by what looks to be a similar amount when you take into account the accumulation length.

jaycedowell, 2023-02-21:
Negative missing data fractions became an error condition with https://github.com/lwa-project/ovro_data_recorder/commit/8b4fce79db74fa7fb3ec961b1ef3e8fb4281551b.

caseyjlaw, 2023-02-21:
That's helpful and I think we found a configuration change to avoid this issue. Closing...

closed 2023-02-21 by caseyjlaw