ovro-lwa/lwa-issues #247: Errors when configuring x-engine

iveydavis, 2023-02-12:
Nikita and I were going to try taking simultaneous beamformed + slow vis data. We were following the system use recipe; the settings we used for the system were whatever are defined in `./scripts/restoreSettings.sh `  and we used the config file in the `lwa_config_calim.yaml`. Pipelines started up okay. We ran into an issue when configuring the x-engine. We tried:
* Just configuring for slow vis
*  Just configuring for a beam with calibration
*  Just configuring for a beam _without_ calibration
*  Configuring for slow vis and a calibrated beam simultaneously
And consistently ran into issues. When configuring for slow vis, we get a sync_time key error:
[image]

when configuring for a beam, we get a chan0 key error:
[image]

yupinghuang, 2023-02-12:
https://github.com/ovro-lwa/lwa-issues/issues/162
https://github.com/ovro-lwa/lwa-issues/issues/210
potentially related.

I think it may be that the GPU nodes aren't getting any packets from the F engine. `like_bmon.py` on lxdlwagpu01 shows no traffic (and spot check a few other).

yupinghuang, 2023-02-12:
[image]
dashboard shows an cross corr accumulation length of 0 for SNAP1. It may be related.

yupinghuang, 2023-02-12:
Yeah both of these errors are key errors from get_bifrost_status. A null check or something and then throwing a "hey the F engine is not talking to me" would be more helpful.

Reinitialized the F engines and configure_xengine went ok.

caseyjlaw, 2023-02-15:
I added a try/except to catch these two errors and log a suggested solution.

closed 2023-02-15 by caseyjlaw