ovro-lwa/lwa-issues #248: LWA_dashboard gets HTTP 500

yupinghuang, 2023-02-12:


caseyjlaw, 2023-02-13:
Seems that some service restarts triggered an error of "Too many open files". Prior to that, it seems etcd was down.
Restarted service with ` sudo systemctl restart dashboard` and all is well.

yupinghuang, 2023-02-18:
It just did that to me again.

caseyjlaw, 2023-02-18:
Ok, I restarted it, but will reopen this until I find a fix.

yupinghuang, 2023-02-19:
I think this happens when I refresh the page too much (partly my bad). It's probably some open file not getting closed while panel hasn't shut down the python process yet. Running `lsof` or something like that to list open file handlers while refreshing a lot should give us a clue of which file it is.

caseyjlaw, 2023-02-20:
I can reproduce this failure when reloading the dashboard several times. `lsof` produces a big list and I'm not sure how to interpret it. Many paths to python shared-object libraries. Many "Permission denied" to /proc files.
Any suggestions for paring this down?

yupinghuang, 2023-02-21:
```
Feb 21 18:49:20 lxdlwacr dashboard[30096]: handle: <Handle BaseAsyncIOLoop._handle_events(4, 1)>
Feb 21 18:49:20 lxdlwacr dashboard[30096]: Traceback (most recent call last):
Feb 21 18:49:20 lxdlwacr dashboard[30096]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/asyncio/events.py", line 145, in _run
Feb 21 18:49:20 lxdlwacr dashboard[30096]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/tornado/platform/asyncio.py", line 189, in _handle_events
Feb 21 18:49:20 lxdlwacr dashboard[30096]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/tornado/netutil.py", line 266, in accept_handler
Feb 21 18:49:20 lxdlwacr dashboard[30096]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/socket.py", line 205, in accept
Feb 21 18:49:20 lxdlwacr dashboard[30096]: OSError: [Errno 24] Too many open files
```

yupinghuang, 2023-02-21:
See if messing with session reaping helps
https://panel.holoviz.org/user_guide/Server_Configuration.html

`--unused-session-lifetime` and `--check-unused-sessions`.

caseyjlaw, 2023-02-21:
I set it to `--unused-session-lifetime 10000` and set the service file to restart more aggressively.
I do see it crashing and restarting. I do not see it hanging on a 500 error.

rh-codebase, 2023-02-21:
restarting more aggressively is not a fix... Well, not a long term fix. Exhausting resources on a host may lead to other processes which need those resources to also fail.

yupinghuang, 2023-02-22:
So panel actually never errors out because it's tries to be nice. It just returns 500. So systemctl restart does not help. 

It seems like this is a situation where a transient DNS resolution failure in the etcd client resulting in potentially connections not getting cleaned up since, probably, the web server still keeps references to the mcs objects.

```
Feb 22 02:21:24 lxdlwacr dashboard[7193]: File "client.py", line 32, in _translate_exception:
Feb 22 02:21:24 lxdlwacr dashboard[7193]:  Traceback (most recent call last):
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/client.py", line 46, in handler
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/client.py", line 570, in add_watch_callback
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/watch.py", line 105, in add_callback
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/watch.py", line 128, in _run
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/grpc/_channel.py", line 426, in __next__
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/grpc/_channel.py", line 809, in _next
Feb 22 02:21:24 lxdlwacr dashboard[7193]: grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
Feb 22 02:21:24 lxdlwacr dashboard[7193]:         status = StatusCode.UNAVAILABLE
Feb 22 02:21:24 lxdlwacr dashboard[7193]:         details = "DNS resolution failed for service: etcdv3service:2379"
Feb 22 02:21:24 lxdlwacr dashboard[7193]:         debug_error_string = "{"created":"@1677032484.807138449","description":"Resolver transient failure","file":"src/core/ext/filters/client_channel/client_channel.cc
Feb 22 02:21:24 lxdlwacr dashboard[7193]: >
Feb 22 02:21:24 lxdlwacr dashboard[7193]: During handling of the above exception, another exception occurred:
Feb 22 02:21:24 lxdlwacr dashboard[7193]: Traceback (most recent call last):
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/bokeh/application/handlers/code_runner.py", line 197, in run
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/proj/lwa-shell/mnc_python/notebooks/LWA_dashboard.ipynb", line 34, in <module>
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/mnc/control.py", line 38, in __init__
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/mnc/control.py", line 99, in set_properties
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/lwa352_pipeline_control/lwa352_pipeline_control.py", line 82, in __init__
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/lwa352_pipeline_control/lwa352_pipeline_control.py", line 348, in __init__
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/lwa352_pipeline_control/lwa352_pipeline_control.py", line 378, in check_connection
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/lwa352_pipeline_control/etcd_control.py", line 269, in send_command
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/client.py", line 48, in handler
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/client.py", line 32, in _translate_exception
Feb 22 02:21:24 lxdlwacr dashboard[7193]: etcd3.exceptions.ConnectionFailedError: etcd connection failed
Feb 22 02:21:24 lxdlwacr dashboard[7193]:  
Feb 22 02:21:24 lxdlwacr dashboard[7193]: 2023-02-22 02:21:24,818 Uncaught exception GET /LWA_dashboard (10.40.0.4)
Feb 22 02:21:24 lxdlwacr dashboard[7193]: HTTPServerRequest(protocol='http', host='localhost:5006', method='GET', uri='/LWA_dashboard', version='HTTP/1.1', remote_ip='10.40.0.4')
Feb 22 02:21:24 lxdlwacr dashboard[7193]: Traceback (most recent call last):
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/tornado/web.py", line 1704, in _execute
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/panel/io/server.py", line 213, in get
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/panel/io/server.py", line 150, in server_html_page_for_session
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/panel/io/resources.py", line 114, in bundle_resources
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/panel/io/resources.py", line 177, in css_raw
Feb 22 02:21:24 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/panel/io/resources.py", line 75, in loading_css
[... cut ...]
Feb 22 02:40:20 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/multiprocessing/context.py", line 67, in Lock
Feb 22 02:40:20 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/multiprocessing/synchronize.py", line 162, in __init__
Feb 22 02:40:20 lxdlwacr dashboard[7193]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/multiprocessing/synchronize.py", line 59, in __init__
Feb 22 02:40:20 lxdlwacr dashboard[7193]: OSError: [Errno 24] Too many open files
```

yupinghuang, 2023-02-22:
So, a temporary fix to try out is...

whenever an etcd3 client is instantiated in the notebook, surround the code block that does stuff with a try catch block. Assume that the function stack never gets cleaned up when an exception is thrown.
```
try:
    do_stuf
except Exception as e:
    # do the cleanup, say del(con) if controller is instantiated
    raise e
```
This will hopefully make it a recoverable HTTP 500.

And you just kinda gotta pray that `from lwa_antpos.reading`, `lwa_antpos.mapping` don't run into errors/somehow gets cleaned up.

Should probably set a `LimitNOFILE` lower than the user limit in the systemd config file too, so that it doesn't interfere with other things running as the same user.

In general, we should use as few etcd clients as possible (it's just good to keep track of connections but also https://grpc.io/docs/guides/performance/). This involves using singleton and also maybe using the same wrapper class (ref https://github.com/ovro-lwa/lwa-issues/issues/260).

caseyjlaw, 2023-02-22:
Shouldn't this be the job of `dsa_store`? Rick developed that layer of code abstraction to control the interface to etcd.

yupinghuang, 2023-02-22:
It's complicated.

First of all, the user is responsible for handling exception that a module raises, otherwise it wouldn't be an exception.

`dsa_store` should retry on transient errors. That's for sure.

But almost nothing is using `dsa_store` in our current setup (correlator control, mcs.client, snap etcd client, lwa_antpos). Waiting for all of them to migrate over for the dashboard to work seems not worth it.

Beyond that, there are cleanup cases where `dsa_store` wouldn't be able to handle. Say you get connection timeout when `put` or something. The most pythonic thing to do is probably to have a context manager, but then that doesn't work with all the objects that possess an etcd client.

So..knowing that the web server has the potential of becoming a zombie process, cleaning up helps.

caseyjlaw, 2023-02-22:
Fair point. We can do both. Adding a try/except in the health dashboard and in `dsa_store` (and migrate from direct etcd to `dsa_store`).
I'll start on the notebook and see if it helps.

caseyjlaw, 2023-03-17:
We're doing better with this, from what I can tell. Closing for now.

closed 2023-03-17 by caseyjlaw