ovro-lwa/lwa-issues #250: WebUI access signup -- collect users

rh-codebase, 2023-02-13:
collect list of users to add to the authDB for the webUI

yupinghuang, 2023-02-14:
yuping

marinanderson, 2023-02-14:
mmanders

caseyjlaw, 2023-02-14:
claw

iveydavis, 2023-02-14:
idavis

rh-codebase, 2023-02-15:
above users added. please add a comment to this issue after successful signup and login so this can be closed.

gregghallinan, 2023-02-15:
gh

rh-codebase, 2023-02-15:
added gh.

ldaddario, 2023-02-15:
ldaddario

jack-h, 2023-02-15:
jackh

nkosogor, 2023-02-15:
nkosogor

iveydavis, 2023-02-15:
Was able to successfully sign in.  Is there going to be a timeout for how long a user can be in control? If someone closes out of the UI window but forgets to log out so that no one else can log in, I can see this becoming an issue (which is to say that I tried logging in a second time to make sure it actually worked but couldn't because idavis (me) was logged in).

rh-codebase, 2023-02-15:
There's currently a 1 hour timeout. I'll add this as a reminder on the page. Another feature would be time remaining next to the Current User on the main landing page. CARMA allowed users to usurp control which can also be implemented and perhaps a more favorable approach.

caseyjlaw, 2023-05-31:
I tried to log in to the webUI authentication, but failed. Can you confirm that I have an account there?

yupinghuang, 2023-05-31:
I can’t either. I think the accts were wiped after the disk failure

rh-codebase, 2023-06-02:
all users readded and need to do a one-time sign-up, then sign-in.