ovro-lwa/lwa-issues #253: Beamforming observing tutorial

caseyjlaw, 2023-02-15:
We have many observing modes, but setup, observing, data reduction can be complex. A document (on SharePoint to start) should describe the steps to set up, observe, and reduce beamformed data.

jaycedowell, 2023-02-21:
One thing that I will point out is that I've been working a simple beam scheduler called [`run_sdf.py`](https://github.com/lwa-project/ovro_data_recorder/blob/main/run_sdf.py).  It's basically a compatibility layer that allows session definition files (SDFs) that we use here in New Mexico to be run at OVRO.  These SDFs follow the structure outlined in our [MCS0030 memo](https://leo.phys.unm.edu/~lwa/lwavt/doc/MCS0030v5_Observing.pdf) although `run_sdf.py` doesn't support all of the beamforming modes listed in that memo.

To use the script you need to:
 1) Create an SDF[^1] - either manually with a text editor or through something like [`sessionGUI.py`](https://github.com/lwa-project/session_schedules/blob/main/sessionGUI.py).
 2) Setup the telescope so that all of the beams you are interested in using are calibrated and that the data recorders are receiving beam data.
 3) Run your SDF by calling `run_sdf.py` with the filename as the only argument at least one minute before the observations start.

The observations should run and you should end up with a `.tgz` file at the end that contains what was run, what file was recorded, and basic information about the telescope setup.

For an example of how SDFs can be used there is our [stress test rpo.](https://github.com/lwa-project/stress_tests) that we use to generate beam cut observations for looking at the beam pointing/SEFD.

[^1]:  There is one cavet about the SDFs - if you want a power beam then you need to set the "spectrometer" keywords in the SDF.  `run_sdf.py` will use these to set the integration time/polarization products on the power beam recording.  Without the "spectrometer" keywords you end up with a voltage beam that requires additional setup to work.

jaycedowell, 2023-02-21:
There looks to also be a version of [`run_sdf.py` in `mnc_python`](https://github.com/ovro-lwa/mnc_python/blob/main/scripts/run_sdf.py) but it looks to be out of date.  These two versions need to merge at some point, probably with the `mnc_python` one becoming the default.

caseyjlaw, 2023-03-08:
Would you be willing to submit a pull request to update `run_sdf.py` in mnc_python?

jaycedowell, 2023-03-29:
`run_sdf.py` was updated in `mnc_python` with https://github.com/ovro-lwa/mnc_python/pull/21

iveydavis, 2023-05-16:
A link to the tutorial btw: https://caltech.sharepoint.com/sites/ovro/projects/SitePages/Beamforming-Tutorial.aspx

closed 2023-10-04 by caseyjlaw