ovro-lwa/lwa-issues #254: Log aggregation

yupinghuang, 2023-02-15:
rsyslog or fluentd or something like that.

@rh-codebase Was there infra from DSA-110? What's a good node for keeping all the logs? If you don't have a good candidate I might just pick a calim node...

rh-codebase, 2023-02-15:
syslog-ng was the daemon being wrapped by the python class for DSA110(repo: git@github.com:dsa110/dsa110-pyutils.git) ./dsautils/dsa_syslog.py. syslog-ng has parsers for JSON formatted logs as well as allowing forwarding of logs off the local host to some central server. 
JSON formatted logs are brutal for human consumption and really meant to have something ingesting them to display to the human. The evaluation of such a tool is still WIP. 

No machine was bought on DSA or LWA for hosting the central logs. Other requirements like depth of logs also needs discussing and documented.

yupinghuang, 2023-03-07:
@rh-codebase any chance I can run a container on the MCS node (or somewhere else) to test some log aggregation?

rh-codebase, 2023-03-07:
all processes on mcs will have some impact on etcd. There's only 77G free on mcs so storing logs there doesn't seem like a good long term solution. feel free to experiment though.

yupinghuang, 2023-03-10:
I set up a test Elasticsearch+Fluentd+Kibana stack on lwacalim10 (https://github.com/ovro-lwa/efk) running podman under root. Lots more config to do as things may just die at any time. But the elasticsearch/fluentd endpoint is lwacalim10:24224 and kibana `lwacalim10:5601`.  The simplest thing one can do it probably install `calyptia-fluentd` and then have it tail some log file and send it over to `lwacalim10:24224` and see how they show up on kibana.
https://docs.fluentd.org/installation/install-by-rpm

There's also a systemd plugin https://github.com/fluent-plugin-systemd/fluent-plugin-systemd. and it support syslog out of the box too.

The other thing is that for the elasticsearch/fluentd instance, one should probably set up a "file" output so that things get written to files and archived on calim10.

Fluentd in principle can convert between different log formats and then send the logs in logstash format to elasticsearch which indexes the logs and then kibana is where you can do fancy (sometimes confusing) things with them.

[image]
(from my test yesterday on calim10. The indexing is still kinda weird and it still doesn't do json quite right and the keys are weird but hey it look pretty)

Anyway, if anyone is interested-- contributions welcome! If things break we can always just restart all the containers,

cc @jack-h

caseyjlaw, 2023-03-10:
Very nice! This has been a long open issue.
I tunneled to lwacalim10:5601 and see kibana, but no data. Any tips?

caseyjlaw, 2023-03-29:
Centralized OVRO server may be good place for this. Kemby working on it.

yupinghuang, 2023-11-26:
So the snap2 etcd, all the systemd logs on calim09, calim10, and /var/log/messages are being ingested into elasticsearch/kibana.

Formatting/indexing still needs bit of work cuz they're in a giant pile. But filtering by `@log_name` is a reasonable start.

I actually quite like keeping the snap2 logs separate so let's not stick it in systemd. Keeping application logs separate makes it easier to parse the formats (when there are lots more bells and whistles in syslog/systemd log).

Note that there's a few-minute latency for log file ingestion -- it's not intended for real-time analysis.

[image]

The fluentd configuration is where the bulk of the work happens , for example https://github.com/ovro-lwa/efk/blob/main/client-conf/calyptia-fluentd-calim10.conf

caseyjlaw, 2023-11-28:
This looks great!
The service was working, but trying again just now does not load the app. Not sure what's wrong.

caseyjlaw, 2023-11-28:
Here's a [query URL](http://localhost:5601/app/discover#/view/35556180-8e06-11ee-b33b-ebc8ff223e69?_g=(filters%3A!()%2CrefreshInterval%3A(pause%3A!t%2Cvalue%3A0)%2Ctime%3A(from%3Anow%2Fd%2Cto%3Anow%2Fd))) for logging from today with `no-os-fluff`:

caseyjlaw, 2023-11-29:
It looks like the lwacalim10 logging is not available on elastic.
For example, I can filter out all logging messages with this query:
`not @log_name snap* and not @logname fluent* and not @log_name lwacalim09 and not @log_name lwastormds`

Shouldn't we have some with `@logname lwacalim10`?

yupinghuang, 2023-11-30:
All the storage logs are now in. Added this because I noticed lustre client eviction messages on them so it'd be good to keep track.

[image]

caseyjlaw, 2023-11-30:
This is great! I'm playing with queries and saved a couple kml queries. I also see some saved at the top. I'm not sure which is more useful for use by the team, but it would be nice to set queries up for OoD and others to get one-click insight.

yupinghuang, 2023-11-30:
Here's a status report

all systemd logs on calim09 and calim10 and lwastor00 are in via logname `lwacalim10` and `lwacalim09` and `lwastormds`.

dr logs are tailed and under logname `dr-*` (for example `dr-data03.pipeline.log.dr-vfast-6.lwacalim03.log`) I haven't figured out a way to shorten that.

snap2-etcd log files are tailed and under logname `snap-` (e.g. `snap.home.pipeline.logs.lwa-feng-etcdsnap07.out`).

storage oss nodes kernel logs are under their host names `lwastor01` etc (with a lot of filtering and we should only see lustre and kernel related things) .

Am I forgetting anything?

caseyjlaw, 2023-12-14:
This is working well. Instructions at https://github.com/ovro-lwa/calim-ops/wiki/2-Using-Kibana-for-Logs.

closed 2023-12-14 by caseyjlaw