ovro-lwa/lwa-issues #255: snap1 vs snap01 in etcd keys

yupinghuang, 2023-02-17:
This came from the dashboard and the Fengine status reported by `control.py` making no sense https://github.com/ovro-lwa/mnc_python/blob/main/mnc/control.py#L190.

etcd timestamp shows that `snap/01` has been the key of choice since...Feb 13?

`ls` is lwastore.

I think all of the single digit SNAP boards are impacted.

```
In [25]: d = ls.get_dict('/mon/snap/1')

In [26]: d['stats']['fpga']['flash_firmware']
Out[26]: 'snap2_f_200msps_64i_4096c_2022-12-15_2103.fpg'

In [27]: d = ls.get_dict('/mon/snap/01')

In [28]: d['stats']['fpga']['flash_firmware']
Out[28]: 'snap2_f_200msps_64i_4096c_2023-02-16_1000.fpg'
```

yupinghuang, 2023-02-17:
oh ok https://github.com/realtimeradio/caltech-lwa/commit/2b8f84a706b07cff7dc16d86549a97d3ad41a32a

yupinghuang, 2023-02-17:
Can we....get rid of the old keys in etcd?

rh-codebase, 2023-02-17:
The DsaStore class exposes delete(<key>). There's an example in the class docs.

yupinghuang, 2023-02-17:
Thanks! will do.

closed 2023-02-17 by yupinghuang