ovro-lwa/lwa-issues #256: snap/corr monitoring points in influxdb

yupinghuang, 2023-02-17:
Rick, is this gonna be a lot of work?

rh-codebase, 2023-02-17:
I started this service a few weeks back. A bit of work.

caseyjlaw, 2023-02-17:
@greghell is developing code to get a few key ARX/F-eng monitor points into influx/grafana. These are specific to the question of RFI saturation, so not all snap/corr is required.
@yupinghuang, do you want all monitor points?

rh-codebase, 2023-02-17:
ARX? All ARX monitor points are already in Influx/Grafana.  I'm working on some of the F-eng.

rh-codebase, 2023-02-17:
I thought Gregg was working on RFI flagging type monitor points.

caseyjlaw, 2023-02-17:
@rh-codebase The ARX power and f-eng FFT overflow are key RFI indicators. I'm helping him on the latter measurement. If you are working on getting all f-eng points into grafana, than he should stop his development. After your work, we'd just need to create the grafana panel with the RFI-relevant measurements. Does that sound right?

rh-codebase, 2023-02-17:
That sounds right. Have the monitor points been identified? The F-eng ones would help me prioritize which ones first. Since logical mp's map to Influx tables, they don't have to go in all at the same time. I can get the ones related to Greg's work first. All the ARX's are in so it's just F-eng and anything Greg may generate from his algorithms. If Greg is generating MPs, then which model does he plan to use to expose them?  push or pull?  The antenna is push on 1sec cadence but ARX is pull and the cadence is controlled via the webUI. F-eng will follow the ARX paradigm unless there's reasons not to.

rh-codebase, 2023-02-17:
Right. The json is fairly deep. thx.

greghell, 2023-02-21:
@rh-codebase would that help if I gave you the snap2_fengine.Snap2FengineEtcd commands I use to read monitoring points?
So far we have (in ETCD format):
* 'pfb', 'get_overflow_count' (per SNAP)
* 'eq', 'clip_count' (per SNAP)
* 'input', 'get_bit_stats' (per antenna/pol)
* 'autocorr', 'get_new_spectra' (per antenna/pol)

I will also need the (I am not certain of the command):
* 'mask' , 'get_acc_len' (per SNAP)
* 'mask' , 'get_flag_threshold' (per antenna/pol)
* 'mask' , 'get_flag_count' (per antenna/pol)

And the ARX config data : lwa_arx.ARX().get_all_chan_cfg

rh-codebase, 2023-02-21:
IIRC I was using one command which dumped all of this in one big JSON packet. Maybe there are other command which break this out into smaller segments. In any case, it won't hurt. 
All the ARX monitor points are in Influx and Grafana. It just needs to be turned on which I believe Casey did this morning.

rh-codebase, 2023-02-21:
Oh. at what cadence do you need this info?

greghell, 2023-02-21:
Great! Right now I am monitoring every minute.

yupinghuang, 2023-02-22:
https://github.com/ovro-lwa/lwa-issues/issues/242

There's X engine capture block throughtput and F engine throughput ['eth'] as well per @jack-h 

So if I'm understanding this correctly, we're sending an etcd command to get status and then have the response sent back and then put parts of the response in etcd?

jack-h, 2023-02-22:
The `etcd` service which handles commands already polls all SNAP status fields and throws them in the etcd `/mon/snap/XX` key every few seconds (maybe ~minute), so you don't actually have to do anything to get them.
The mapping from the commands Greg lists is straightforward -- if you JSON decode the monitor key, there will be a `mask` key which is another hash containing the values Greg mentions. Essentially, anything returned by `Snap2Fengine.get_status_all()` is already in etcd.

The same is true of the X-engine stats -- the `get_status` commands issued by pipeline control instances just grab things from etcd which are already being pushed there. In the case of the correlator the key path is `/mon/corr/<gpu hostname>/pipeline/<pipeline ID>/<blockname>`

Eg. I think the capture stats will be in `/mon/corr/<hostname>/pipeline/[0|1|2|3]/udp_verbs_capture` which is a dictionary which should include a `gbps` entry. (the capture block is a bit of a special case, since it's name is determined by bifrost, not me)

yupinghuang, 2023-02-22:
rip I kept trying to get `/mon/snap/0x/fpga` to no avail

jack-h, 2023-02-22:
Yar, it's not keyed like that -- `/mon/snap/01` contains _everything_. Would be easy to slightly change things so this isn't the case, and the first level of the dictionary could be turned into an etcd key name

rh-codebase, 2023-02-22:
the cadence control needs to be handled by either jack's service or the one I'm working on. If it's mine, then it needs to poll the service instead of getting stuff out of etcd to guarantee the interval is correct. If Jack's service handles it, then my service just has to set a watch for changes to /mon/snap/xx.  I'm putting the control in my service since I already had code which handles the cadence command. I'm happy to just set a 'watch'.

jack-h, 2023-02-22:
on the SNAPs etcd gets updated every minute, for the X-engines it's about every 10 seconds

jack-h, 2023-02-22:
The SNAP service can be polled to retrieve the equivalent data directly, the X-engine one cant

rh-codebase, 2023-02-25:
@jack-h I decided to grab from etcd to keep f-eng and X-engine symmetric. Can f-eng push at 1 sec the stuff needed at that rate and the rest can come at 1 min?

jack-h, 2023-02-25:
> Alas it cannot

If you list what you want at what cadence I can try to implement that.
But, the etcd interface stores everything in one big key (I actually think
it didn't used to, but was changed so as to lower the etcd server load) so
there isn't the facility to timestamp different bits of information
differently. The easiest thing would be to turn off all the monitoring for
blocks you don't care about, and then run the loop at a high time cadence.

caseyjlaw, 2023-02-25:
I think cadences of 1m on f-eng and 10s on x-eng are a good start. We should probably have a direct chat about etcd structure and design (issue #243 tracks that).

rh-codebase, 2023-02-27:
i should be able to have most of this in grafana today.

rh-codebase, 2023-02-27:
the following are being inserted into Influx:
host, overflow_count, fft_shift, clip_count, acc_len, rms00-63, flag_threshold00-63

A few Grafana plots are in the Snap dashboard.  Still WIP

rh-codebase, 2023-03-04:
influxdb be dragons. carry big stick

appears influxdb v1.7 retains knowledge of measurement names even after dropping them from the database resulting in what it thinks are duplicate column names. This forces queries to append ::tag or ::field to keys not to mention the confusion of why the tables contain columns which are no longer being inserted. If this all sounds confusing, it is.

workaround: drop the measurement and create another one with a different name and just live with it.

The good news is this new approach is working and allows a more natural way to define monitor points as arrays and assign tags to them.

rh-codebase, 2023-06-09:
Closing out. Open new issue if other monitor points are needed and other related changes.

closed 2023-06-09 by rh-codebase