ovro-lwa/lwa-issues #257: zombie xengine processes?

yupinghuang, 2023-02-20:
`like_bmon.py` didn't show any bifrost process, but the `lwa352-pipeline` processes are still alive.

[image]

This suggests that the last metric was sent was Feb 17 16:40ish which was around the time https://github.com/ovro-lwa/lwa-issues/issues/248 happened.

here's some log at the time, which may or may not be helpful
```
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27806/Beamform/perf", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27806/BeamformVlbiOutput/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27804/Beamform/perf", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27804/BeamformVlbiOutput/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27807/BeamformSumBeams/perf", "w") failedproclog.cpp
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: :245 internal error: fopen("/dev/shm/bifrost/27805/BeamformSumBeams/perf", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27806/BeamformSumBeams/perf", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27804/BeamformSumBeams/perf", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27806/BeamformOutput/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27804/BeamformOutput/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27807/BeamformOutput/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27805/BeamformOutput/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27805/Beamform/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27805/Corr/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27804/Corr/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: ring_impl.cpp:223 Condition failed: !_nwrite_open
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: ring_impl.cpp:223 error 11: BF_STATUS_INVALID_STATE
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: ring_impl.cppring.cpp::223119 Condition failed:  error !_nwrite_open11
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: : ring_impl.cppBF_STATUS_INVALID_STATE:
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: 223 error 11: BF_STATUS_INVALID_STATE
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cppring.cpp::119 error 24511 internal error: : fopen("/dev/shm/bifrost/27807/Beamform/stats", "w") failedBF_STATUS_INVALID_STATE
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27806/Corr/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: ring_impl.cpp:223 Condition failed: !_nwrite_open
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: ring_impl.cpp:223 error 11: BF_STATUS_INVALID_STATE
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: ring.cpp:119 error 11: BF_STATUS_INVALID_STATE
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27807/Corr/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: ring_impl.cpp:223 Condition failed: !_nwrite_open
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: ring_impl.cpp:223 error 11: BF_STATUS_INVALID_STATE
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: ring.cpp:119 error 11: BF_STATUS_INVALID_STATE
Feb 18 00:44:54 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cppproclog.cpp::245245 internal error:  internal error: fopen("/dev/shm/bifrost/27805/Copy/perf", "w") failedfopen("/dev/shm/bifrost/2
Feb 18 00:44:54 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27807/Copy/perf", "w") failed245 internal error:
Feb 18 00:44:54 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: fopen("/dev/shm/bifrost/27806/Copy/perf", "w") failed
```

jack-h, 2023-02-21:
Hmmmm. Not sure I have any wisdom here unless this can be recreated. Expanding around the logs:

```
Feb 18 00:42:42 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Breaking from packet receive because of so many late packets
Feb 18 00:42:42 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Breaking from packet receive because of so many late packets
Feb 18 00:42:42 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Breaking from packet receive because of so many late packets
Feb 18 00:42:42 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Breaking from packet receive because of so many late packets
Feb 18 00:43:19 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Found device 0: NVIDIA GeForce RTX 2080 Ti
Feb 18 00:43:19 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Found device 1: NVIDIA GeForce RTX 2080 Ti
Feb 18 00:43:19 lxdlwagpu02 systemd-journald[700]: Suppressed 8445 messages from lwa-xeng-etcd.service
Feb 18 00:43:19 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Using device 0: NVIDIA GeForce RTX 2080 Ti
Feb 18 00:43:19 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: IP is: 1107306762 (10.41.0.66)
Feb 18 00:43:19 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Found device 0: NVIDIA GeForce RTX 2080 Ti
Feb 18 00:43:19 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Found device 1: NVIDIA GeForce RTX 2080 Ti
Feb 18 00:43:19 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Using device 1: NVIDIA GeForce RTX 2080 Ti
Feb 18 00:43:19 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: IP is: 285223178 (10.41.0.17)
Feb 18 00:43:20 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Found device 0: NVIDIA GeForce RTX 2080 Ti
Feb 18 00:43:20 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Found device 1: NVIDIA GeForce RTX 2080 Ti
Feb 18 00:43:20 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Using device 0: NVIDIA GeForce RTX 2080 Ti
Feb 18 00:43:20 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: IP is: 1107306762 (10.41.0.66)
Feb 18 00:43:20 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Found device 0: NVIDIA GeForce RTX 2080 Ti
Feb 18 00:43:20 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Found device 1: NVIDIA GeForce RTX 2080 Ti
Feb 18 00:43:20 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: Using device 1: NVIDIA GeForce RTX 2080 Ti
Feb 18 00:43:20 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: IP is: 285223178 (10.41.0.17)
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27806/Beamform/perf", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27806/BeamformVlbiOutput/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27804/Beamform/perf", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27804/BeamformVlbiOutput/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27807/BeamformSumBeams/perf", "w") failedproclog.cpp
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: :245 internal error: fopen("/dev/shm/bifrost/27805/BeamformSumBeams/perf", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27806/BeamformSumBeams/perf", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27804/BeamformSumBeams/perf", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27806/BeamformOutput/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27804/BeamformOutput/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27807/BeamformOutput/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27805/BeamformOutput/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27805/Beamform/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27805/Corr/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: proclog.cpp:245 internal error: fopen("/dev/shm/bifrost/27804/Corr/stats", "w") failed
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: ring_impl.cpp:223 Condition failed: !_nwrite_open
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: ring_impl.cpp:223 error 11: BF_STATUS_INVALID_STATE
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: ring_impl.cppring.cpp::223119 Condition failed:  error !_nwrite_open11
Feb 18 00:44:53 lxdlwagpu02 lwa352_start_xeng_etcd_service.py[9711]: : ring_impl.cppBF_STATUS_INVALID_STATE:
[... cut ...]
Feb 20 21:11:36 lxdlwagpu02 python[9711]: 2023-02-20 21:11:36,003 - /home/ubuntu/proj/lwa-shell/caltech-bifrost-dsp/pipeline-control/scripts/lwa352_start_xeng_etcd_service.py - ERROR - No PIDfile (/home/ubuntu/xpipeline.0.pid) available
```

So, something (probably F-engine restarts or other upstream misbehaviour) broke the packet receiver. Then it looks like the pipeline tried to start again, and everything went wrong. What is surprising / suspicious is that the pipeline was never commanded to stop before restarting.

jack-h, 2023-02-21:
Not sure if it is relevant that GPU-02 is the only machine that has been provisioned for triggered buffering to be functional. This may have increased the likelihood that someone (me!) was poking around outside the main system control and could have broken something. Seems like it should have broken long before this issue though

jaycedowell, 2023-02-21:
Reprovisioning - that reminds me - Mark Haynes has some data on gpu09 from the 2010 XC15 experiment that shouldn't get blown away.  @caseyjlaw or @rh-codebase do either of you know if he's been able to access/download those data?

caseyjlaw, 2023-02-21:
I do not know, but gpu09 is treated as a backup node and supports some ad hoc analysis. It won't get reprovisioned without notification, but we should get his data off so we have the freedom of a reprov.

yupinghuang, 2023-02-21:
Looks like GPU02 does it again...and fairly close in time to dashboard giving up too...

jack-h, 2023-02-21:
The dashboard only interacts with `etcd`? Does it only watch keys, or does it actually trigger things on the GPU hosts?

jack-h, 2023-02-21:
(I assume you are implying that the GPU node dying is killing the dashboard, not the dashboard is killing the GPU node(?!))

yupinghuang, 2023-02-21:
Oh I'm assuming dashboard dying (or something that it's doing related to it dying) is killing the GPU node. The dashboard runs on lxdlwacr. Ludicrous I know. More to come later...

jack-h, 2023-02-21:
Hahaha. I'll eagerly await updates.

rh-codebase, 2023-02-21:
@jaycedowell I also don't know whether Mark Haynes has accessed his data.

marinanderson, 2023-02-22:
Confirming that Mark Haynes has not yet downloaded the data. @rh-codebase, he will reach out to you in the next day or so with an ssh key and some instructions on how to access the data.

Can we also back it up somewhere else? Calim node?

jack-h, 2023-03-08:
@yupinghuang Any updates on this and/or its relation to dashboard failure?

yupinghuang, 2023-03-08:
No no clue at al. The most recent time I saw a similar error message was during https://github.com/ovro-lwa/lwa-issues/issues/280 but none of that `fopen('/dev/shm/)` stuff.

The stuff with `fopen('/dev/shm')` make me think it might have to do with the OS deciding that it needs the RAM back for some reason (either someone called `shm_open` or the OS really needed RAM..)