ovro-lwa/lwa-issues #260: Define process to better dissiminate software to co-developers

rh-codebase, 2023-02-21:
My impression is that the SW meetings are insufficient to convey what functionality already exists for other developers to use resulting in duplication. We have been using various forms of the following. This list is likely incomplete:

- verbal via SW meetings
- README.md in the respos
- Written docs stored in Sharepoint.

This issue is to start a discussion of how and where this information lives and how everyone can find  it.

yupinghuang, 2023-02-21:
docstrings on https://ovro-lwa.github.io/

rh-codebase, 2023-02-21:
We're re-using SW from DSA so it must also be made aware of. This issue was born out of insufficient knowledge of a class wrapping etcd which comes from DSA-110 origins. There will likely be SW in ovro-caltech(meant to emphasize SW for all projects) which needs broader exposure to everyone writing SW. The myrepos framework is an attempt to accomplish some of this by allowing SW devs to pull down all relevant repos for a specific project no matter where they reside. More is needed since I suspect the README's in each repo aren't sufficient to allow one to determine if SW in it meets their requirements. And last but not least, SW from NM is starting to be reused here.

caseyjlaw, 2023-03-08:
At today's meeting, we set expectations for developers to use `lwa-shell` and the `swdeploy` mechanism to get it deployed. We'll continue to implement that, but as far as this issue goes, the process is more clear.
I'll close this issue, but reopen if we're failing at the implementation.

caseyjlaw, 2023-03-09:
Totally fair. I've reopened this issue.

rh-codebase, 2023-03-09:
No. We don't want N implementations talking to etcd. What we want is a single module/class/entry point which wraps etcd. What if we swap out etcd? Or need to scale it horizontally? It will be a nightmare to update N implementations. Wrapping is the way forward. What requirements are needed for the wrapper API?

yupinghuang, 2023-03-14:
WARNING: long-term thing that's not currently helpful

https://fastapi.tiangolo.com/features/

Not sure where things like this fit in but I've been thinking for a while now -- having a nice framework for defining a web service that automatically generates clients in different languages and does dependency injection (ok some people hate it but I like it) can be nice.

For the command bit of MNC, we are having a client talk to a service (DsaStoreService/etcd) which has a bunch of other control services (which possess DsaStoreClient) registering callbacks on the other side and executing commands. I do think that the control services should have access to a reliable key value store, but I wonder if sending commands needs to necessarily go through that key-value store. Having the control services define that own RESTful API (which prevents chaos and bloodshed) but then have the DsaStore focus on the server side functionalities may make life easier for everyone...

for the next project....

rh-codebase, 2023-03-14:
A direct command to service should be looked at. The one 'pro' for using the DKVS is that every dev can look at another's code and fairly quickly understand the architecture. We could attempt to standardize the direct approach but then it's 2 paradigms. Not a killer but I do feel that the quicker we can understand the other's code the better. This comes into play during code reviews. We want those fast but not at the expense of glossing over bad patterns.