ovro-lwa/lwa-issues #261: DNS resolution failed for service: etcdv3service:2379

yupinghuang, 2023-02-21:
```
Feb 21 18:48:32 lxdlwacr dashboard[30096]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/client.py", line 46, in handler
Feb 21 18:48:32 lxdlwacr dashboard[30096]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/client.py", line 570, in add_watch_callback
Feb 21 18:48:32 lxdlwacr dashboard[30096]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/watch.py", line 105, in add_callback
Feb 21 18:48:32 lxdlwacr dashboard[30096]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/watch.py", line 128, in _run
Feb 21 18:48:32 lxdlwacr dashboard[30096]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/grpc/_channel.py", line 426, in __next__
Feb 21 18:48:32 lxdlwacr dashboard[30096]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/grpc/_channel.py", line 809, in _next
Feb 21 18:48:32 lxdlwacr dashboard[30096]: grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
Feb 21 18:48:32 lxdlwacr dashboard[30096]:         status = StatusCode.UNAVAILABLE
Feb 21 18:48:32 lxdlwacr dashboard[30096]:         details = "DNS resolution failed for service: etcdv3service:2379"
Feb 21 18:48:32 lxdlwacr dashboard[30096]:         debug_error_string = "{"created":"@1677005312.658297703","description":"Resolver transient failure","file":"src/core/ext/filters/client_channel/client_channel.c
Feb 21 18:48:32 lxdlwacr dashboard[30096]: >

rh-codebase, 2023-02-21:
That's a start. lots of things could cause this so we're not at root cause yet. a quick work-around would be to add an entry for etcdv3service in /etc/hosts on lxdlwacr. If the error occurs after that, then the problem is likely local to lxdlwacr. If it was some sort of DoS on DNS, then more than this service would be throwing errors. If it is local to lxdlwacr then possibly a buggy python implementation in the etcd client code.

yupinghuang, 2023-02-22:
I can see icmp failures on grafana for cr and the gpu nodes. I mean, this just looks like a situation where the client has gotta retry

[image]

rh-codebase, 2023-02-22:
that seems a bit odd. Something must be causing those. a client retry would certainly help but would be nice to dig deeper. Maybe when Morgan powers off the shelter for his work we can bring stuff up slowly and check these things out.

yupinghuang, 2023-05-17:
https://github.com/lwa-project/ovro_data_recorder/issues/33

this seems to be happening more often  lately.

caseyjlaw, 2023-08-10:
This hasn't recurred for a while and we no longer run much on lxdlwacr. I'll close unless there are objections.

closed 2023-08-10 by caseyjlaw