ovro-lwa/lwa-issues #262: hello again slow etcd?

yupinghuang, 2023-02-23:
this started with Jack's correlator clients check_connection failed (aka Connection failed ref'd in https://github.com/ovro-lwa/lwa-issues/issues/153)

yupinghuang, 2023-02-23:
[image]
[image]

yupinghuang, 2023-02-23:
[image]

rh-codebase, 2023-02-24:
#153 turned out to be a system problem from a ubuntu sw update. It looks like LXD was again updated 2 days ago. I've tried disabling snap updates but apparently that method no longer works. I'll be taking the mcs host down so all m/c will be down.

caseyjlaw, 2023-03-08:
Maybe related from Jack today:
```
I had issues initializing over etcd, I think possibly related to the current rapid polling of various monitor points going on. I worked around this by disabling the etcd services, initializing the boards, then turning the services back on.
Loading settings over etcd seemed to work without issue.
```

Looking at the etcd info in grafana, I see many raft proposals and more than 13000 keys. Is that a sign of a problem?

yupinghuang, 2023-03-09:
Here's more https://github.com/ovro-lwa/lwa-issues/issues/290

yupinghuang, 2023-05-17:
[image]

ouch

rh-codebase, 2023-06-09:
Pretty sure this is related to and being track in #92  So no work to be done here until #92 is closed and slowness occurs after that. Removing In Progress.

caseyjlaw, 2023-07-21:
etcd slowness may have caused more problems yesterday, but the connection is unclear.
There was certainly a coincidence between etcd high sync duration and failures when configuring the f-engines via etcd. Using a direct f-engine access seemed to work better. Today, the etcd delay is down and the systems restored themselves.

Here's the sync duration plot for the last day. 
[image]