ovro-lwa/lwa-issues #263: Calim 40G traffic unintentional broadcast

jack-h, 2023-02-23:
Yesterday I noticed a bunch of traffic for `10.41.0.76` (calim) ending up in the wrong place (gpu-01). 

I've just looked at the observation config file and I see that this is the destination of beam data on the calim nodes. Do the calim nodes periodically announce their presence to the 40G switch (the GPU nodes have a cronjob to ping occasionally) and/or should their MAC addresses be fixed in the switch's mac address table?

yupinghuang, 2023-02-23:
Does the packet get broadcasted if the switch is aware of the calim nodes, but calim isn't listening?

jack-h, 2023-02-23:
Nope. Broadcast basically means that the switch doesn't know where the
traffic should go, which I think implies the destination hasn't sent any
recent (where "recent" depends on switch settings) traffic which would make
the switch aware of it's existence.

rh-codebase, 2023-02-24:
I don't see any cron jobs on calim01.  Regarding the GPU nodes, one of the pings is now incorrect. @jack-h it looks like those pings were set up to ping the broadcast address. yes? When one of the 10.41.0.xx nets was expanded to a /26 the bcast became 10.41.0.127. The ping is still hitting 10.41.0.47.  This may  not really matter much.

I've never seen a need for this type of band-aid so does everyone agree that this is due to a bad switch? If so, why not replace the broken switch?

rh-codebase, 2023-02-24:
added cronjob to all calim nodes under user: ubuntu

5,15,25,35,45,55 * * * * /bin/ping 10.41.0.127 -b -c 1 -w 1

closed 2023-02-24 by rh-codebase

jack-h, 2023-02-24:
I see this all the time with interfaces that are only used for high speed
RX and generate no outbound traffic. If no traffic is ever generated by a
server interface, the switch to which it is connected can't learn that it
is there. Possibly there is also something on the switch which can generate
traffic specifically to overcome this issue.
I'm not sure why ARP requests from the transmitting servers (in this case
the GPU nodes) aren't generating traffic from the receive side -- I don't
know what rules the transmitter uses to figure out how often to send
requests to a destination to which it is constantly sending.

For the GPU nodes, the MAC/switch port mappings are statically set on the
switch (probably why the now faulty cronjob doesn't matter). It would be
easy to do a similar thing for Calim, it just means annoyingly having to
gather all the relevant MAC address information together (and changing
things when hardware moves). I'm sure there is a solution which is both
better than the static mac address-table and generating traffic via ping,
but I don't know what it is.

rh-codebase, 2023-02-24:
I don 't either but I too thought ARP should be sufficient. It's how all other discovery takes place. When you said you noticed this on all RX high speed interfaces, were those interfaces also embedded inside a class C network?  I just can't get my brain to accept this config.

jack-h, 2023-02-24:
Hmmm, good question. I guess in my life I invariably have an FPGA at one
end, and a GPU at the other. The FPGAs don't ARP (you have to tell them the
destination MACs of the place they are transmitting to), so the GPU servers
live in the shadows and generally never transmit anything at all. In these
situations the broadcast problem is pretty common.
I guess I've never seen this in a situation where the traffic is
originating from a CPU/GPU server, where the transmitter must be using ARP
to get the MAC of the receiver.

rh-codebase, 2023-02-24:
thanks for the reminder about FPGA's not ARPing. Seems like a nice feature to add if possible. Thoughts of adding a thread to the GPU code to do a ping like command? That way it's self-contained and not hanging out in a cron jobs.

jack-h, 2023-02-25:
> thanks for the reminder about FPGA's not ARPing. Seems like a nice feature
> to add if possible.

Yeah, it's kind of a big feature though!

Thoughts of adding a thread to the GPU code to do a ping like command? That
> way it's self-contained and not hanging out in a cron jobs.
Could do, but you can't really guarantee that anything is running faster
than the base OS. I suppose there could be a service just doing this.
But, I think my preference might be to have a service on the cosmic ray
node which sends periodic ARP requests to everyone on the 40G net. That
would, at least, mean there is only one service running in one place