ovro-lwa/lwa-issues #264: Beamformer WriterOp died

yupinghuang, 2023-02-24:
```
Feb 24 04:35:14 lwacalim01 dr-beam-2[2325710]: Exception in thread WriterOp:
Feb 24 04:35:14 lwacalim01 dr-beam-2[2325710]: Traceback (most recent call last):
Feb 24 04:35:14 lwacalim01 dr-beam-2[2325710]:  File "/home/pipeline/.conda/envs/datarecorder/lib/python3.6/threading.py", line 916, in _bootstrap_inner
Feb 24 04:35:14 lwacalim01 dr-beam-2[2325710]:    self.run()
Feb 24 04:35:14 lwacalim01 dr-beam-2[2325710]:  File "/home/pipeline/.conda/envs/datarecorder/lib/python3.6/threading.py", line 864, in run
Feb 24 04:35:14 lwacalim01 dr-beam-2[2325710]:    self._target(*self._args, **self._kwargs)
Feb 24 04:35:14 lwacalim01 dr-beam-2[2325710]:  File "/home/pipeline/ovro_data_recorder/dr_beam.py", line 467, in main
Feb 24 04:35:14 lwacalim01 dr-beam-2[2325710]:    active_op.write(time_tag, idata)
Feb 24 04:35:14 lwacalim01 dr-beam-2[2325710]:  File "/home/pipeline/ovro_data_recorder/filewriter.py", line 235, in write
Feb 24 04:35:14 lwacalim01 dr-beam-2[2325710]:    data = self.reduction(data)
Feb 24 04:35:14 lwacalim01 dr-beam-2[2325710]:  File "/home/pipeline/ovro_data_recorder/reductions.py", line 45, in __call__
Feb 24 04:35:14 lwacalim01 dr-beam-2[2325710]:    odata = self._average(idata)
Feb 24 04:35:14 lwacalim01 dr-beam-2[2325710]:  File "/home/pipeline/ovro_data_recorder/reductions.py", line 37, in _average
Feb 24 04:35:14 lwacalim01 dr-beam-2[2325710]:    odata = idata.reshape(-1,self.time_avg,idata.shape[1],idata.shape[2],idata.shape[3])
Feb 24 04:35:14 lwacalim01 dr-beam-2[2325710]: ValueError: cannot reshape array of size 12288000 into shape (1024,1,3072,4)
```

I think this may be when there's packet loss?

yupinghuang, 2023-02-24:
for the record I was trying to do a 2000ms time averaging

yupinghuang, 2023-02-24:
it'd make sense if it was trying to do 1000ms as opposed to 1024ms averaging????

jaycedowell, 2023-02-24:
With the current configuration of the recorders everything is done in units of 1024 spectra so any averaging in time has to be a clean divisor of that.  I went with powers-of-two because it is easy to explain.

There's nothing that says that it is required that the spectra are worked on in sets of 1024.  1000 would work with allowed averaging times of something like 1, 5, 10, 50, 100, 500, or 1000 spectra.  1000 gets you an integration time of 1003.1 ms.

jaycedowell, 2023-02-24:
https://github.com/lwa-project/ovro_data_recorder/commit/f3c2c5bb25b9d81d4017b411d8a403228a5b3b94 add better input validation to the HDF5 record command.

yupinghuang, 2023-02-24:
https://github.com/ovro-lwa/mnc_python/blob/main/mnc/control.py#L372 so the control code apparently sent the command for 1024 ms.

jaycedowell, 2023-02-24:
Huh, so it did:
```
2023-02-24 04:35:14 [INFO    ] Started operation - <HDF5Writer filename='/data01/beam01/059999_043513451819982bc69',
    start_time='2023-02-24 04:35:13.451819', stop_time='2023-02-24
    15:41:53.451819', reduction=<ReductionOperation time_avg=1024,
    chan_avg=1, pols='XX,YY,CR,CI'>>
```

caseyjlaw, 2023-02-24:
yes, the control code converts to meet the power of 2 requirement. logging should confirm that.

jaycedowell, 2023-02-24:
I spend all this time talking about 1024 and find out that the default in `dr_beam.py` is 1000.  Classic.

jaycedowell, 2023-02-24:
https://github.com/lwa-project/ovro_data_recorder/commit/01b5a5882efdc15fa97c6a4f2b77be82eb064046 - it really is 1024 now.

closed 2023-03-02 by yupinghuang