ovro-lwa/lwa-issues #265: last system config update did not create new system_cfg.yml file

rh-codebase, 2023-02-24:
@yupinghuang committed an updated System Config excel file to the repo, committed, pushed and reloaded via the webui but something failed to produce a new system_cfg.yml file.

rh-codebase, 2023-02-24:
unit test using the new System Config excel file passed by successfully creating system_cfg.yml. Need time to see if this issue can be reproduced

caseyjlaw, 2023-04-21:
I tried this again and the webUI does not seem to work.
I have a new version of the antenna spreadsheet in github. I logged in, clicked "reload configuration", and waited/reloaded the page. No change.
The configuration is about 1 month old and does not include several repaired antennas, so it would be nice to get this working again.

caseyjlaw, 2023-06-06:
I see that there is a recent update to `/cfg/system`, so I think there was progress on this. Could you update in this thread?
The structure of the info at that key are different from what is expected by `lwa-antpos` code, so I'd like to understand how it got pushed in.

rh-codebase, 2023-06-09:
Successfully parsed and pushed configs via WebUI. Removing  "In Progress"