ovro-lwa/lwa-issues #269: 2023/03/01 Beamforming Issues Report

iveydavis, 2023-03-01:
I ran into a few different issues while beamforming this afternoon, so maybe it would be better to split this across multiple issues. But for the sake of continuity for when the different issues arose, I figured it be good to document them all here.

First thing: the original goal of the of the observation was to confirm that the `time_avg` setting for `con.start_dr()` was now working after pulling the latest version of the mnc repo.

Second thing: I didn't run the restoreSettings.sh script as is documented in the System Use Recipe since Gregg had set up the day time settings earlier in the day. I don't think this would be responsible for the errors I ran into though. 

Because the original goal was just to check that `time_avg` setting was now working for values other than 1ms, I didn't bother calibrating beam 1 and I didn't bother with tracking:
```
con = control.Controller('config/lwa_config_calim.yaml')
con.configure_xengine(['dr1'])
con.control_bf(num = 1, targetname = 'zenith', track = False)
con.start_dr(recorders = ['dr1'], duration = 10*1e3, time_avg = 128)
```

In my endless folly, I forgot to check the status of the data recorders before doing this-- the error of only 8 of 9 threads being active was still persisting from a previous beamforming attempt. Weirdly enough though, it seems to have still produced a file. I restarted dr1 and recorded again, this time with a 60s observation and longer integration time: `con.start_dr(recorders = ['dr1'], duration = 60*1e3, time_avg = int(2**9))` and all seemed to go well. 

I wanted to check on how a calibrated vs uncalibrated beam might compare since we have new system settings, so in the same terminal as the previous commands, I ran: `con.configure_xengine(recorders = ['dr2'], calibratebeams = True)`, the output of which included a few interesting responses. The full output is attached below.
[calibrate_beams_output.txt](https://github.com/ovro-lwa/lwa-issues/files/10865981/calibrate_beams_output.txt)
A summary of the errors:
-  Several pipelines failed to respond to etcd command
-  Several watch callback failures due to an Attribute error (`AttributeError: 'RevisionCompactedError' object has no attribute 'events'`) that precedes a pipeline that fails to respond to etcd command
- lxdlwagpu04 and lxdlwagpu06 are the only hosts that didn't run into these issues as far as I can tell
-  The first 4 pipelines (first 2 subbands?) weren't calibrated (I think this is consistent with the latest cal tables though)
-  Only one pipeline was found for the last subband?
- It took nearly 25 minutes to calibrate a single beam (not an error, but unfortunate)

After beam 2 was done calibrating, I tried resetting the pointing for beam 1 (`con.control_bf(num = 1, targetname = 'zenith', track = False)`), but apparently configuring for beam 2 reset the configuration for beam 1, and I got the following attached error:

[image]

oh well. I set the pointing for beam 2 and still try to record with two beams:
  ```
con.control_bf(num = 2, targetname = 'zenith', track = False)`
con.start_dr(recorders = ['dr1','dr2'], duration = 60 * 1e3, time_avg = int(2**9))
```
but __beam 2__ ran into the "Only 8 of 9 threads active" error. After the recording, both data recorder logs showed a queue size of 1 for the data recorders, so I reset both data recorders.

Once calim is back up, I can post pictures of the data that were produced.

yupinghuang, 2023-03-05:
"but apparently configuring for beam 2 reset the configuration for beam 1"

Yeah. https://github.com/ovro-lwa/mnc_python/blob/main/mnc/control.py#L219 https://github.com/ovro-lwa/mnc_python/blob/main/mnc/control.py#L224

It really doesn't need to do that though. It also interferes with all other modes of observing

jaycedowell, 2023-03-06:
The `self.bfc.clear()` is there because restarting the x-engine pipelines clears all beam calibration information.

caseyjlaw, 2023-03-06:
We set up an option to configure the x-engine with a second terminal without resetting or clearing configuration. I'll close this, unless others object.

closed 2023-03-06 by caseyjlaw