ovro-lwa/lwa-issues #272: mnc.control.Controller start_fengine doesn't work on etcd

yupinghuang, 2023-03-02:
The timestamps are super sus. Reprogramming does not run that fast. Also the ERRORs.

It did not reprogram snap03, which was on an older firmware. It did reinitialized all the boards, I think. But snap02 stopped sending packets after that.

```
In [26]: con.start_fengine(initialize=True, program=True, force=True)
2023-03-02 18:29:36,540 - lwa_f.blocks.block:snap01 - WARNING - eth - Couldn't find Ethernet core. Will retry later
2023-03-02 18:29:36,574 - mnc.control - INFO - snap01 is already programmed.
2023-03-02 18:29:36,596 - mnc.control - INFO - Starting f-engine on snap01
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
2023-03-02 18:29:46,612 - Snap2FengineEtcd - ERROR - Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
2023-03-02 18:29:56,631 - Snap2FengineEtcd - ERROR - Received 0 of 1 expected responses
2023-03-02 18:29:56,635 - lwa_f.blocks.block:snap02 - WARNING - eth - Couldn't find Ethernet core. Will retry later
2023-03-02 18:29:56,658 - mnc.control - INFO - snap02 is already programmed.
2023-03-02 18:29:56,680 - mnc.control - INFO - Starting f-engine on snap02
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
2023-03-02 18:30:06,696 - Snap2FengineEtcd - ERROR - Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
2023-03-02 18:30:16,715 - Snap2FengineEtcd - ERROR - Received 0 of 1 expected responses
2023-03-02 18:30:16,721 - lwa_f.blocks.block:snap03 - WARNING - eth - Couldn't find Ethernet core. Will retry later
2023-03-02 18:30:16,745 - mnc.control - INFO - snap03 is already programmed.
2023-03-02 18:30:16,767 - mnc.control - INFO - Starting f-engine on snap03
Received 0 of 1 expected responses
[... 341 lines cut ...]
2023-03-02 18:33:07,543 - Snap2FengineEtcd - ERROR - Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
Received 0 of 1 expected responses
2023-03-02 18:33:17,562 - Snap2FengineEtcd - ERROR - Received 0 of 1 expected responses
```

yupinghuang, 2023-03-02:
Also this is far less information than what the user needs to figure out whether the configuration succeeded and where the issue was.

yupinghuang, 2023-04-04:
shouldn't it be path to a config file living on the cr node?

jack-h, 2023-04-04:
<these last 2 comments meant for issue #273 >

caseyjlaw, 2023-04-07:
The config file does indeed need to be visible to the pipeline user.
I've removed the `useetcd` option and aligned the f-engine start procedure with the script used by Jack.

caseyjlaw, 2023-04-07:
Most of those timeouts were related to not using the timeout argument well. That's better now.

closed 2023-04-07 by caseyjlaw