ovro-lwa/lwa-issues #273: start_fengine(useetcd=False) throws exception

yupinghuang, 2023-03-02:
```
In [30]: con.start_fengine(initialize=True, program=True, force=True, useetcd=False)
2023-03-02 18:56:06,194 - lwa_f.blocks.block:snap01 - INFO - adc - Detected FMC ADC board on port 0
2023-03-02 18:56:06,951 - lwa_f.blocks.block:snap01 - INFO - adc - Detected FMC ADC board on port 1
2023-03-02 18:56:06,951 - lwa_f.blocks.block:snap01 - WARNING - powermon - Error while trying to initialize I2C objects
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-30-8d16a2da067f> in <module>()
----> 1 con.start_fengine(initialize=True, program=True, force=True, useetcd=False)

~/.conda/envs/datarecorder/lib/python3.6/site-packages/mnc/control.py in start_fengine(self, initialize, program, force, useetcd)
    149                 f = snap2_fengine.Snap2Fengine(snap2name)
    150 
--> 151             if program and f.fpga.is_programmed():
    152                 logger.info(f'{snap2name} is already programmed.')
    153 

~/.conda/envs/datarecorder/lib/python3.6/site-packages/lwa_f/snap2_fengine.py in <lambda>(*args, **kwargs)
    763         return val
    764     else:
--> 765         return lambda *args, **kwargs: object.__getattribute__(self, '_intercept_method_call')(self.blockname, attr, val, args, kwargs)
    766 
    767 def _intercept_method_call(self, blockname, methodname, method, args, kwargs):

~/.conda/envs/datarecorder/lib/python3.6/site-packages/lwa_f/snap2_fengine.py in __getattribute__(self, attr)
    757     elif attr == "intercept_method_call":
    758         wrap_method = False
--> 759     val = object.__getattribute__(self, attr)
    760     if not callable(val):
    761         return val

AttributeError: 'Fpga' object has no attribute 'blockname'
```

caseyjlaw, 2023-04-04:
(Sorry, I posted my query in the wrong issue a moment ago. This is the right one)
@jack-h Is this syntax still supported?

```
f = snap2_fengine.Snap2Fengine('snap02')
f.cold_start_from_config(config_file='config/lwa_config_calim.yaml')
```

If I run this, I get an error:
`AttributeError: 'Adc' object has no attribute 'blockname'`

jack-h, 2023-04-04:
Are you running the latest `lwa_f` library?

jack-h, 2023-04-04:
Also, to @yupinghuang 's comment on issue 272, the path to the config file should be appropriate so as to be interpreted by the etcd service on `lwa-cr`

jack-h, 2023-04-06:
After some random experimenting, I can recreate this if I instantiate `Snap2FengineEtcd` and also `Snap2Fengine`. I'll try and figure out why this is, (but I'd also like to claim this is just unsupported operation).

caseyjlaw, 2023-04-06:
I am inclined to leave it be and we design `mnc-python` to be an etcd-only design. Direct snap control can be documented as via the `caltech-lwa` control code.

caseyjlaw, 2023-04-07:
I've removed the `useetcd` option and aligned the f-engine start procedure with the script used by Jack.
Tests are looking good, so I'll close this and merge with main.

closed 2023-04-07 by caseyjlaw