ovro-lwa/lwa-issues #275: ERROR - adc - FMC 0: Data calibration Failure!

yupinghuang, 2023-03-02:
Remembered this happening before but it seems like it might have retried? It's hard to tell because it logged an error and then moved on and redid the
```
INFO - adc - Chip 0, Lane 0:    XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX---|----XXXXXXXXXXXXXXXXXXXXXXXX
```
thing and then moved on.

yupinghuang, 2023-03-02:
looks like it doesn't retry. the next block is for FMC1.

yupinghuang, 2023-03-05:
Ok consistently see these. All other FMCs seem fine
```
2023-03-05 17:17:45,561 - lwa_f.blocks.block:snap01 - ERROR - adc - FMC 1: Data calibration Failure!
```

```
2023-03-05 17:26:37,307 - lwa_f.blocks.block:snap02 - ERROR - adc - FMC 0: Data calibration Failure!
```

jack-h, 2023-03-08:
Snap02 looks like a genuine hardware fault --
[image]

yupinghuang, 2023-03-08:
How do I tell? Is it that... Chp 4 Lane 7 has no vertical bar in it?

(I'm sure there's a doc somewhere that says something about what those outputs mean...feel free to just paste the link)

jack-h, 2023-03-08:
Sorry, no docs for this. The columns are delay trials as the firmware tries to find a delay which, when applied to the ADC data signals, gives rise to reliable data capture. Horizontal lines represent individual wires between ADC chip and FPGA.
Symbol meanings are:
"X": data was corrupt at the trialed delay.
"-" :  data was OK at the trialed delay
"|" :  the firmware chose this delay as the best available.

All X means data were never able to be captured reliably.
Working configurations look like the other rows -- a solid section of "-", with the midpoint marked with "|"

jack-h, 2023-03-31:
Single lane faults like this can sometimes be cured by re-seating the ADC/FMC cards on the problem SNAP2. If this can't be solved, the single problem lane will manifest as glitches in the ADC time stream for a single analog input.

yupinghuang, 2023-09-23:
still failing. May be a good time to plan on doing the reseating thing.

yupinghuang, 2023-10-11:
whoa, chip 4 lane 7 calibrated successully by a razor thin margin today

```
2023-10-11 14:19:39,779 - lwa_f.blocks.block:snap02 - INFO - adc - Chip 4, Lane 7:    XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX|XXXXXXXXXXXXXXXXXXXXXXXX
```