ovro-lwa/lwa-issues #277: ARX service probably didn't come up after power down

yupinghuang, 2023-03-02:
```
In [3]: ma = lwa_arx.ARX()
/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/lwautils/conf/etcdConfig.yml

In [4]: dds = ma.get_all_chan_cfg(11)
---------------------------------------------------------------------------
ServiceNoResponseException                Traceback (most recent call last)
~/anaconda3/envs/casa/lib/python3.6/site-packages/lwautils/cmd_rsp.py in send(self, key, cmd, val, timeout)
    170                 if self.response_d is None:
--> 171                     raise snre.ServiceNoResponseException()
    172                 self.my_store.delete(self.full_response_key)

ServiceNoResponseException: 

During handling of the above exception, another exception occurred:

ServiceNoResponseException                Traceback (most recent call last)
<ipython-input-4-4e1373b2921a> in <module>
----> 1 dds = ma.get_all_chan_cfg(11)

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwautils/lwa_arx.py in get_all_chan_cfg(self, arx_addr, user_timeout)
    926 
    927         """
--> 928         chan_cfgs = self._get_all_chan_cfg(arx_addr, user_timeout)
    929         #self._show_chan_cfg(0, chan_cfg)
    930         rtn = []

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwautils/lwa_arx.py in _get_all_chan_cfg(self, arx_addr, user_timeout)
   1040         """
   1041 
-> 1042         rtn = self._send(arx_addr, 'geta', '', user_timeout)
   1043 
   1044         self._check_rtn(rtn, SET_CHAN_CFG_ERRORS)

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwautils/lwa_arx.py in _send(self, arx_addr, cmd, val, user_timeout)
    386 
    387         cmd_key = '{}{}'.format(CMD_KEY_BASE, arx_addr)
--> 388         rtn = self.my_cr.send(cmd_key, cmd, val, user_timeout)
    389 
    390         return rtn

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwautils/cmd_rsp.py in send(self, key, cmd, val, timeout)
    174             except:
    175                 # Oops, no response from service and user_timeout reached
--> 176                 raise snre.ServiceNoResponseException()
    177         else:
    178             self.my_store.delete(self.full_response_key)

ServiceNoResponseException: 
```

rh-codebase, 2023-03-02:
i'll take a look after comap meeting.

rh-codebase, 2023-03-02:
service is up

closed 2023-03-02 by rh-codebase