ovro-lwa/lwa-issues #278: lwa352_start_xeng_etcd_service.py[5611]: Failed to create queue pair: (1) Operation not permitted

yupinghuang, 2023-03-02:
```
Mar 02 21:54:58 lxdlwagpu03 lwa352_start_xeng_etcd_service.py[5611]: terminate called after throwing an instance of 'Verbs::Error'
Mar 02 21:54:58 lxdlwagpu03 lwa352_start_xeng_etcd_service.py[5611]:   what():  Failed to create queue pair: (1) Operation not permitted
Mar 02 21:54:58 lxdlwagpu03 lwa352_start_xeng_etcd_service.py[5611]: terminate called after throwing an instance of 'Verbs::Error'
Mar 02 21:54:58 lxdlwagpu03 lwa352_start_xeng_etcd_service.py[5611]:   what():  Failed to create queue pair: (1) Operation not permitted
Mar 02 21:54:58 lxdlwagpu03 lwa352_start_xeng_etcd_service.py[5611]: terminate called after throwing an instance of 'Verbs::Error'
Mar 02 21:54:58 lxdlwagpu03 lwa352_start_xeng_etcd_service.py[5611]:   what():  Failed to create queue pair: (1) Operation not permitted
Mar 02 21:54:59 lxdlwagpu03 lwa352_start_xeng_etcd_service.py[5611]: terminate called after throwing an instance of 'Verbs::Error'
Mar 02 21:54:59 lxdlwagpu03 lwa352_start_xeng_etcd_service.py[5611]:   what():  Failed to create queue pair: (1) Operation not permitted
```

yupinghuang, 2023-03-02:
um, well, I guess everything is fine. It's just tell me that I tried to set a pipeline's output when it  doesn't exist?

yupinghuang, 2023-03-03:
Oh wait this was an ibd issue https://github.com/ovro-lwa/lwa-issues/issues/89

Maybe translate this to a hint

jack-h, 2023-03-08:
I thought the necessary restart of the driver in issue #89 was implemented somewhere? I guess this is not the case @rh-codebase

rh-codebase, 2023-03-08:
The systemd service script doesn't seem to work. See #89. I do believe systemd runs the script but somethings else is failing.