ovro-lwa/lwa-issues #280: lxdlwagpu03 lxd.daemon[7118]: => LXD failed to start

yupinghuang, 2023-03-03:
and it keeps retrying and failing...doesn't look good?

yupinghuang, 2023-03-03:
[loggy.log](https://github.com/ovro-lwa/lwa-issues/files/10885144/loggy.log)
looks like this thing is only on some nodes and not others (gpu01 doesn't seem to have it).

03, 05, 07 are the only nodes showing these log entries and they also seem to be the nodes which I am having trouble sending xengine commands to

rh-codebase, 2023-03-03:
one can use:  sudo snap remove lxd
on hosts which do not require it.
lxd has been removed on lxdlwagpu03, 05,07

closed 2023-03-03 by rh-codebase

yupinghuang, 2023-03-03:
Why were they there? The nodes didn't get reprovisioned after reboot,
correct?

rh-codebase, 2023-03-03:
early on, gpu nodes were to be config. like dsa nodes but was then abandoned. Just left over cruft. The nodes do not get reprovisioned on reboot.