ovro-lwa/lwa-issues #281: calim06 can't talk to any 10.40 addresses

yupinghuang, 2023-03-03:
```
(base) [pipeline@lwacalim00 ~]$ ssh ubuntu@10.41.0.81
ubuntu@10.41.0.81's password: 
Activate the web console with: systemctl enable --now cockpit.socket

Register this system with Red Hat Insights: insights-client --register
Create an account or view all your systems at https://red.ht/insights-dashboard
Last login: Thu Mar  2 06:01:14 2023 from 10.40.0.42
Could not chdir to home directory /home/ubuntu: No such file or directory
[ubuntu@lwacalim06 /]$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp68s0f0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 3c:ec:ef:61:3a:26 brd ff:ff:ff:ff:ff:ff
    inet 10.40.0.32/24 brd 10.40.0.255 scope global noprefixroute enp68s0f0
       valid_lft forever preferred_lft forever
    inet6 fe80::3eec:efff:fe61:3a26/64 scope link tentative noprefixroute 
       valid_lft forever preferred_lft forever
3: enp68s0f1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 3c:ec:ef:61:3a:27 brd ff:ff:ff:ff:ff:ff
4: enp193s0f0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 40:a6:b7:5a:6e:d0 brd ff:ff:ff:ff:ff:ff
5: enp193s0f1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 9000 qdisc mq state UP group default qlen 1000
    link/ether 40:a6:b7:5a:6e:d1 brd ff:ff:ff:ff:ff:ff
    inet 10.41.0.81/26 brd 10.41.0.127 scope global noprefixroute enp193s0f1
       valid_lft forever preferred_lft forever
    inet6 fe80::42a6:b7ff:fe5a:6ed1/64 scope link 
       valid_lft forever preferred_lft forever
6: virbr0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether 52:54:00:ee:42:df brd ff:ff:ff:ff:ff:ff
    inet 192.168.122.1/24 brd 192.168.122.255 scope global virbr0
       valid_lft forever preferred_lft forever
```
but
```
[ubuntu@lwacalim06 /]$ ping 10.40.0.20
PING 10.40.0.20 (10.40.0.20) 56(84) bytes of data.
From 10.40.0.32 icmp_seq=4 Destination Host Unreachable
From 10.40.0.32 icmp_seq=5 Destination Host Unreachable
From 10.40.0.32 icmp_seq=6 Destination Host Unreachable
^C
--- 10.40.0.20 ping statistics ---
8 packets transmitted, 0 received, +3 errors, 100% packet loss, time 7158ms
pipe 3

```

yupinghuang, 2023-03-07:
Rebooting calim06

yupinghuang, 2023-03-07:
rebooting didn't help with this one `10.40.0.32` is still not there

kembyct, 2023-03-07:
The interface cable was disconnected
system restart NetworkManager. 
The Interface has been back up.

closed 2023-03-07 by kembyct

yupinghuang, 2023-03-08:
Thanks! Did you find the cable disconnected? Or did you disconnected and then reconnected it?

kembyct, 2023-03-08:
It was disconnected. The cable was broken, so I had to replace it. It was up when I tested it again this morning.

rh-codebase, 2023-03-08:
@kembyct The 10.40.0.xx net is the CAT5 interfaces. for lwacalim06 it was not connected to the switch. This happened when the switch was replaced and this cable was just not reconnected to the switch.  The broken fiber cable you are referring to is for the 40GbE on the 10.41.0.xx subnet.

kembyct, 2023-03-08:
Thanks for the clarification,