ovro-lwa/lwa-issues #282: calim04 missing 40 GbE interface

yupinghuang, 2023-03-03:
ref https://github.com/ovro-lwa/lwa-issues/issues/144 and all other prev issues related to this

``
(base) [ubuntu@lwacalim04 ~]$ lspci | grep -i ether
44:00.0 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GBASE-T (rev 02)
44:00.1 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GBASE-T (rev 02)
```

rh-codebase, 2023-03-03:
ref #134  captures prev. steps taken.

yupinghuang, 2023-03-07:
Actually -- gonna try restarting the node real quick

yupinghuang, 2023-03-07:
It's back...Well that was easy
```
(base) [yuping@lwacalim04 ~]$ lspci | grep -i ether
44:00.0 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GBASE-T (rev 02)
44:00.1 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GBASE-T (rev 02)
c1:00.0 Ethernet controller: Intel Corporation Ethernet Controller XL710 for 40GbE QSFP+ (rev 02)
c1:00.1 Ethernet controller: Intel Corporation Ethernet Controller XL710 for 40GbE QSFP+ (rev 02)
```

pinging works

closed 2023-03-07 by yupinghuang