ovro-lwa/lwa-issues #283: 2023-03-05 16:57:48,105 - lwa_f.blocks.block:snap01 - WARNING - sync - System time and GPS time seem to differ by 102 ms

yupinghuang, 2023-03-05:


yupinghuang, 2023-03-05:
```
2023-03-05 16:57:41,095 - lwa_f.blocks.block:snap01 - INFO - adc - Detected FMC ADC board on port 0
2023-03-05 16:57:41,673 - lwa_f.blocks.block:snap01 - INFO - adc - Detected FMC ADC board on port 1
2023-03-05 16:57:47,099 - lwa_f.snap2_fengine:snap01 - INFO - Updating telescope time
2023-03-05 16:57:48,105 - lwa_f.blocks.block:snap01 - WARNING - sync - System time and GPS time seem to differ by 102 ms
2023-03-05 16:57:48,105 - lwa_f.blocks.block:snap01 - INFO - sync - Loading new telescope time at Sun Mar  5 16:57:49 2023
2023-03-05 16:57:52,312 - lwa_f.blocks.block:snap01 - INFO - sync - Detected sync period 536870912.0 (2^29.0) clocks
2023-03-05 16:57:55,028 - lwa_f.blocks.block:snap01 - INFO - sync - Loading new telescope time at Sun Mar  5 16:57:57 2023
2023-03-05 16:57:57,777 - lwa_f.snap2_fengine:snap01 - INFO - Enabling the PFB FIR filter
2023-03-05 16:57:57,784 - lwa_f.snap2_fengine:snap01 - INFO - Disabling EQ TVGs...
2023-03-05 16:57:57,788 - lwa_f.snap2_fengine:snap01 - INFO - Arming sync generators
2023-03-05 16:57:57,933 - lwa_f.blocks.block:snap01 - INFO - packetizer - Full data rate is 50.33 Gbps
2023-03-05 16:57:57,933 - lwa_f.blocks.block:snap01 - INFO - packetizer - 0.95 link occupation => Max 0.75 bandwidth sent
2023-03-05 16:57:57,933 - lwa_f.blocks.block:snap01 - INFO - packetizer - 0.95 link occupation => Max 3092 channels sent
2023-03-05 16:57:57,933 - lwa_f.blocks.block:snap01 - INFO - packetizer - Will allocate 3072 channels in 32 packets
2023-03-05 16:57:57,933 - lwa_f.blocks.block:snap01 - INFO - packetizer - Number of words used: 6208
2023-03-05 16:57:57,933 - lwa_f.blocks.block:snap01 - INFO - packetizer - Number of words spare: 1984
2023-03-05 16:57:57,933 - lwa_f.blocks.block:snap01 - INFO - packetizer - Number of spare chans per packet: 32
2023-03-05 16:57:57,933 - lwa_f.blocks.block:snap01 - INFO - packetizer - Using 24 spare chans per packet
2023-03-05 16:57:59,158 - lwa_f.snap2_fengine:snap01 - INFO - Enabling Ethernet output
2023-03-05 16:57:59,165 - lwa_f.snap2_fengine:snap01 - INFO - Startup of snap01 complete
```

jack-h, 2023-03-08:
Don't think this was reporting anything actually wrong, but was just an unlucky time-register poll cadence. The threshold for reporting a warning is 100ms.
I've increased the poll precision to hopefully avoid this issue occurring in future (https://github.com/realtimeradio/caltech-lwa/commit/ec2c5d3e93e630369be44b76eaf16f5dc0c8a7a5)

yupinghuang, 2023-03-09:
```
2023-03-09 04:53:02,809 - lwa_f.blocks.block:snap03 - WARNING - sync - System time and Sync time seem to differ by 131 ms
```
...?

jack-h, 2023-03-09:
Fighting the urge to just change the tolerance to 200ms

closed 2023-07-28 by jack-h