ovro-lwa/lwa-issues #286: info: Packet loss during receive >10% (75.2% missing)

yupinghuang, 2023-03-07:
Does this just mean...like...75%? calim08

```
2023-03-07 04:35:25 [DEBUG   ] === Status Report ===
2023-03-07 04:35:25 [DEBUG   ]  summary: error
2023-03-07 04:35:25 [DEBUG   ]  info: Packet loss during receive >10% (75.2% missing)
2023-03-07 04:35:25 [DEBUG   ]  queue size: 1
2023-03-07 04:35:25 [DEBUG   ]  active operation: True
2023-03-07 04:35:25 [DEBUG   ]  active filename: slow
2023-03-07 04:35:25 [DEBUG   ]  active time remaining: 364 days, 23:36:05.109262
2023-03-07 04:35:25 [DEBUG   ]  elapsed time: 1.864 s
2023-03-07 04:35:25 [DEBUG   ] ===   ===
```

yupinghuang, 2023-03-07:
ok turns out that storageLogger has been running once per min when I thought I had it set to once per hour...I monkey patched it on calim but forgot to commit it........................

Also Gregg said he was copying files. Gonna keep this open until I try another round of data recording

For the record,

```
2023-03-07 04:12:10 [DEBUG   ]  info: Max. processing time at >125% of gulp (13.366 s = 133.2% )
```

and then moved onto

```
2023-03-07 04:16:41 [DEBUG   ]  info: Packet loss during receive >10% (24.8% missing), Max. processing time at >125% of gulp (13.789 s = 137.5% )
```
intermittently.

```
2023-03-07 04:37:24 [DEBUG   ]  info: Error condition(s) cleared
```

closed 2023-03-29 by caseyjlaw

jaycedowell, 2023-03-29:
Yes, `info: Packet loss during receive >10% (75.2% missing)` means that the packet loss is above 10%, which triggers an error condition.  The actual packet loss when that error condition was last asserted is given in the parentheses.