ovro-lwa/lwa-issues #287: Backup nodes for data recorder

yupinghuang, 2023-03-07:
In case nodes gone down (e.g. right now calim04 and calim04 are having networking issues), can we have a way for the datarecorder to use the other calim nodes (00, 09, 10) for vis data recording?

caseyjlaw, 2023-03-07:
This would require adding it to the configuration yaml and starting dr services on the new nodes. I've done part of the former in a commit in my branch.
Are we comfortable deploying and maintaining recorders on all 11 cal-im nodes? If so, then we would only need to ask users to start them on backup nodes when needed.

yupinghuang, 2023-03-07:
"Are we comfortable deploying and maintaining recorders on all 11 cal-im nodes? "

They share the same home directory so that should be fine. It's the service files that need a bit of finesse. But for the most part I think we should be ok.

jaycedowell, 2023-03-08:
I did a little work on this today with https://github.com/lwa-project/ovro_data_recorder/commit/3e936528238976e685b6a18a55320eb14c79404a.  That commit changed the `generate_services.py` script to take in a JSON configuration file to set the parameters in the systemd service files.  Since everything is currently done as user services that live in the home directory it should be "easy" to move to other nodes as long as the config. file is consistent with the moves.  The moves should also be largely transparent to end users since the visibility managers get updated as part of this and the beam control names are pinned to the beam number rather than the host.

jaycedowell, 2023-12-06:
`generate_services.py` plus the config. file has been working for me for other service changes so I expect it will also work for changing hosts.

closed 2023-12-06 by jaycedowell