ovro-lwa/lwa-issues #290: RevisionCompactedError & Handling

yupinghuang, 2023-03-07:
Been seeing this more and more since etcd's disk sync duration is still sky high (?). It seems like most of these come from the watch functions. (e.g. https://github.com/ovro-lwa/lwa-issues/issues/269). It also showed up when the beamformer tracks a source sometimes.

The clients are not ready to handle this since we're seeing quite a bunch of `AttributeError: 'RevisionCompactedError' object has no attribute 'events'`.

My very naive theory is that the code is watching the `resp` key while another (few ) command was pushed, and then etcd decided to compact the `resp` key and then the watcher got disappointed?

Have a feeling that this is a hot key issue where all the commands for a gpu node (ppipeline?) uses the same cmd key in etcd
https://github.com/realtimeradio/caltech-bifrost-dsp/blob/main/pipeline-control/lwa352_pipeline_control/lwa352_xeng_etcd_client.py#L74. I can be wayyyy off in my theory though.

jack-h, 2023-03-09:
Who is actually in charge of the definition of these interfaces? This seems like a very sensible improvement, but would obviously be client-breaking. If I am responsible for both the service and the client then that's fine, but if this is the case then really there's no standardization of anything in the interface beyond what I arbitrarily choose, at which point it's not really clear to me what etcd is bringing to the table.

caseyjlaw, 2023-03-09:
We have an open issue (surprise!) to define a style guide for subsystems (esp for etcd). I recognize that you raised the question of design and style for M&C many times over the last two years.
In the absence of a style guide, I think it would be good to get one-on-one time with Rick on these kinds of issues.

rh-codebase, 2023-03-09:
A new option could be sent along with the command which shouldn't be client breaking. Clients can choose how they receive their response. 

Just saw Casey's response. I still believe we need to talk as a group on what is available before starting any coding project/feature.

caseyjlaw, 2023-12-14:
Closed via #92.

closed 2023-12-14 by caseyjlaw