ovro-lwa/lwa-issues #294: can't ping lwastor01 from lwamaas

rh-codebase, 2023-03-08:
lwamaas:~# ping lwastor01
PING lwastor01.mcs.pvt (10.40.0.48) 56(84) bytes of data.
From eth1.lwamaas.sas.pvt (10.40.0.4) icmp_seq=1 Destination Host Unreachable

kembyct, 2023-03-09:
The interface has been up

rh-codebase, 2023-03-09:
Something must be inconsistent.

ubuntu@lwamaas:~$ ping lwastor01
PING lwastor01.mcs.pvt (10.40.0.48) 56(84) bytes of data.
From eth1.lwamaas.sas.pvt (10.40.0.4) icmp_seq=1 Destination Host Unreachable
From eth1.lwamaas.sas.pvt (10.40.0.4) icmp_seq=2 Destination Host Unreachable
From eth1.lwamaas.sas.pvt (10.40.0.4) icmp_seq=3 Destination Host Unreachable
From eth1.lwamaas.sas.pvt (10.40.0.4) icmp_seq=4 Destination Host Unreachable
^C

kembyct, 2023-03-13:
The network service has not been loaded. systemctl start network.service fail to start the whenever I tried to run the command.