ovro-lwa/lwa-issues #296: Deployment using myrepo

yupinghuang, 2023-03-08:


yupinghuang, 2023-03-08:
I want a release branch...for rollback reasons

rh-codebase, 2023-03-08:
versions serve the same purpose but that's a detail. Here's a ref. to a framework I tried out:
git@github.com:ovrocaltech/swdeploy.git  Brief summary in the README

caseyjlaw, 2023-03-08:
To be clear, this is the file that defines the code that gets deployed: https://github.com/ovro-lwa/lwa-shell/blob/main/.mrconfig_production.

The `swdeploy` code is what does the deployment and we basically never need to touch it.

Question for @rh-codebase: To be clear, we requiring that every repo at ovro-lwa has a script "deploy" in the base repo directory? This is an example I found: https://github.com/ovrocaltech/deploy-test/blob/main/deploy. Can you confirm?

caseyjlaw, 2023-03-08:
Here is a list of repos that we could start with in myrepos:

- ovro-lwa/lwa-pyutils
- ovro-lwa/lwa-antpos
- ovro-lwa/mnc_python
- realtimeradio/caltech-lwa
- realtimeradio/caltech-bifrost-dsp
- ledatelescope/bifrost
- dsa110/dsa110-pyutils
- ovro-lwa/lwa-cfg
- lwa-project/ovro_data_recorder

yupinghuang, 2023-03-08:
Nope! Let's start with calim since it's lower stake (?)

You can probably create a directory under `/opt` called `/opt/prod-env/` or something and have it owned by pipeline and readonly to all else. And then  wrangle conda to make an env there (https://stackoverflow.com/questions/48871289/how-to-share-an-anaconda-python-environment-between-multiple-users) and then add that path to all users' `.condarc`.

rh-codebase, 2023-03-08:
a file named 'deploy' with execute mode set. Correct, the swdeploy service doesn't need touching. It just runs 'deploy' as a system command in every repo checked out by myrepo

The crux will be if some repo depends on some other repo having some lib installed. I don't attempt to resolve  repo dependencies.

caseyjlaw, 2023-03-08:
The README for `swdeploy` does not describe where the service runs or what key triggers its execution. Can you add that?

yupinghuang, 2023-03-10:
> Here is a list of repos that we could start with in myrepos:
> 
>     * ovro-lwa/lwa-pyutils
> 
>     * ovro-lwa/lwa-antpos
> 
>     * ovro-lwa/mnc_python
> 
>     * realtimeradio/caltech-lwa
> 
>     * realtimeradio/caltech-bifrost-dsp
> 
>     * ledatelescope/bifrost
> 
>     * dsa110/dsa110-pyutils
> 
>     * ovro-lwa/lwa-cfg
> 
>     * lwa-project/ovro_data_recorder
> 
> 
> Any additions?

The deploy script will be different for some of those repos, depending on the kind of node, right? Is it possible for us to have. like

`lwa_calim_shell` and `lwa_gpu_shell` and have different etcd keys they listen too?

caseyjlaw, 2023-03-17:
I've created a few pull requests to move OVRO-LWA code toward automatic deployment. Initial target is python code installed on cal-im nodes.
Some notes:
- setuptools version >=66 requires PEP 440 compliance on version names. caltech-lwa in particular is not compliant and cannot be built. Alternatively installation should use `pip install --use-pep517 .`  @jack-h 
- dsa110-pyutils has a ton of requirements that are not relevant to lwa. It would be nice to restructure that so dsa_store can be used with minimal requirements. Perhaps this could be "ovro_store"? @rh-codebase 
- Did we have an idea for giving observers access to common, difficult to install code (e.g., casacore or f-eng or x-eng control libraries)? Is that done by logging in as "pipeline" or are we asking them to build it? @yupinghuang

yupinghuang, 2023-03-17:
Did we have an idea for giving observers access to common, difficult to install code (e.g., casacore or f-eng or x-eng control libraries)? Is that done by logging in as "pipeline" or are we asking them to build it?

I was thinking
- a shared conda environment across all users (read-only except for the pipeline user) that has the production control stack (put it somewhere in /opt there's like a conda add env thing). And deploy would target that environment. We should save the environment.yml file in the mnc_python repo. This may require getting resource files right too (ref https://setuptools.pypa.io/en/latest/pkg_resources.html) eventually

- If a user is developing on one of the control packages, they could probably use the working directory trick to shadow the installed ones. (or pip install --local --development ????)
- or they can always create their own envs. We bought a few TB SSD for /opt for a reason

rh-codebase, 2023-03-17:
I'd like to see how the general sw community responds to PEP 440. This will become a cluster if each language adopts its own versioning scheme. I'm all for a versioning standard but it must be language agnostic IMO.

I've been planning on  refactoring/moving a DsaStore like class to OVRO general although I don't see in general how one gets around the dependency issues. It will always be there as more code is added to a repo. This is generally the argument of not naming things as 'misc', 'utils', 'common', 'shared', etc as they are too general and lead to such problems.

rh-codebase, 2023-03-17:
@caseyjlaw "lwa_calim_shell and lwa_gpu_shell and have different etcd keys they listen too?" 
The deploy client running on each host would still need to know which scripts to run and this gets a bit messy when changing that information. Can you elaborate more on the concept of lwa_gpu_shell etc. ?

caseyjlaw, 2023-03-17:
@yupinghuang 
> * a shared conda environment across all users

I tested adding the `pipeline` user env path to my .condarc. That makes environments visible if they are in /opt, but the current key ones are in ~pipeline/.conda.
From a bit of testing, I think we could make this work if we rebuild the environments in `pipeline` (this .condarc) is now using /opt.

@rh-codebase 
IMO, that refactor to create `ovrostore` would be very worth while. dsa110-pyutils collected a lot of cruft.

yupinghuang, 2023-03-17:
Rebuilding sounds good. The data recorder env is already a little bloated
btw. Also now that the paths are around you can just cp the env to /opt and
name it something else and it should still work

rh-codebase, 2023-03-17:
And 'utils' will not be any name.

as it sits now, if these deploy scripts are to run on calim nodes only there will need to be checks inside them.

caseyjlaw, 2023-03-26:
> Rebuilding sounds good. The data recorder env is already a little bloated btw. Also now that the paths are around you can just cp the env to /opt and name it something else and it should still work

I actually tried to do this without a rebuild. An `observing` environment is now in /opt and seems to show up for users. I noticed that ipython did not run properly, so I uninstalled and reinstalled it. I'm not convinced the copy worked well, but ipython works now...

caseyjlaw, 2023-03-26:
> as it sits now, if these deploy scripts are to run on calim nodes only there will need to be checks inside them.

I've added a check for cal-im to the `deploy` scripts I set up. It would be great to test this limited set in a deployment from the webUI.

jack-h, 2023-03-28:
I've changed the versioning of `lwa_f` to be PEP440 compliant -- it uses a version scraped from a git tag, so also means you can't install from anything which isn't tagged. Not sure if this is a pro or a con, I'm sure I'll live to regret it at some point.

yupinghuang, 2023-03-29:
Nice!

```
(base) [yuping@lwacalim00 ~]$ conda activate /opt/devel/pipeline/envs/observing
(observing) [yuping@lwacalim00 ~]$ which ipython
/opt/devel/pipeline/envs/observing/bin/ipython
(observing) [yuping@lwacalim00 ~]$ which python
/opt/devel/pipeline/envs/observing/bin/python
(observing) [yuping@lwacalim00 ~]$ ipython
Python 3.6.13 |Anaconda, Inc.| (default, Jun  4 2021, 14:25:59) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.16.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from mnc.control import Control
Read antpos from etcd
Downloading https://hpiers.obspm.fr/iers/bul/bulc/Leap_Second.dat
|=============================================================================================| 1.3k/1.3k (100.00%)         0s
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-1e2d50f77eb2> in <module>
----> 1 from mnc.control import Control

ImportError: cannot import name 'Control'

In [2]: from mnc.control import Controller
```

caseyjlaw, 2023-03-29:
Good reference deploy script: https://github.com/ovro-lwa/lwa-pyutils/blob/main/deploy.

caseyjlaw, 2023-03-29:
I just realized that I left the reference to `datarecorder` in the v1.4.4 tagged version. We decided to migrate code to `deployment` or `development`. I have updated that and bump the version.

caseyjlaw, 2023-03-30:
I've updated the following repos with the new style deploy bash script:
- lwa-pyutils
- lwa-antpos
- mnc_python
- dsa110-pyutils

These have all been tagged and the tags are included in the .mrconfig_production of the lwa-shell.
Once the deploy service is up, we can test these.

jack-h, 2023-03-31:
I've added a deploy script to `lwa_f` and added a tag to `.mrconfig_production`.
Pushed to `lwa-shell:main` (sorry if this wasn't the right thing to do

yupinghuang, 2023-04-12:
What about ovro_data_recorder?

jaycedowell, 2023-04-12:
Maybe?  It would be more involved than that example because you would need to get/build/install Bifrost from the correct version of caltech-lwa-dsp and setup/install the systemd services. The software is also not pip install-able right now.

caseyjlaw, 2023-06-06:
After powering systems back up, we struggled with failures in the x-engine and data recoder setup. It turned out that both systems had old code (`casa` for gpu servers and `datarecorder` on cal-im).
It would be great to get automatic deployment set up at least for the python-only packages listed above.

caseyjlaw, 2023-06-15:
I've made a PR for this new branch.

jack-h, 2023-06-15:
FYI - I've made the `caltech-bifrost-dsp` repo work with `deploy`

caseyjlaw, 2023-06-16:
Thanks, everyone. I'll update the system to use the latest:
```
realtimeradio/caltech-lwa@v2.0.20
realtimeradio/caltech-bifrost-dsp@v1.0.4
lwa-project/ovro_data_recorder@main
ovro-lwa/lwa-pyutils@v1.4.5
ovro-lwa/lwa-antpos@0.6.9
ovro-lwa/mnc_python@0.8.8
ovro-lwa/dsa110-pyutils@v3.8.1
```

These are now set in the `lwa-shell` production config file and tagged as "v7.2.0".
Still waiting to get cal-im to use the deployment, but I tried testing via the WebUI deployment service. The new version is discovered, but deployment seems to fail for me:
[image]

jack-h, 2023-06-16:
Sorry, I should probably know this, but how do I test the deployment
service to recreate this issue? So far I've just been testing by pulling
the tip of the main branch of whatever repository, running the deploy
script manually and observing it complete OK.

caseyjlaw, 2023-06-16:
It could use a dedicated doc, but the "webUI" is a service linked from the ovro-lwa.github.io landing page. It requires a sign up and then sign in to acquire control. Then click on the "deploy" tab to see the interface to that part.

jack-h, 2023-06-16:
Thanks. I see I made a stupid syntax error in the myrepos git commands --
I've fixed this and pushed the new v7.3.0 version.
This seemed to deploy OK, but I don't think the GPU-side services restarted
properly (they do when I just run `deploy` manually). I'll try and debug
this over the weekend

rh-codebase, 2023-06-16:
The failure should only be related to lwa-pyutils. There's more to it than what is being done. This feature is still somewhat WIP. I'll look into why it's failing but the swdeploy config file specifies which repos to run the deploy script in. Currently, only the swdeploy repo and the lwa-pyutils repo. This allow one to specify which host type installs which software and also allow me to test using a test repo w/out side effects. When I get the current failure fixed, which repo should I add to the config as the next repo to deploy?  I'd like to take this 1 step at a time to more easily identify and fix bugs.

caseyjlaw, 2023-06-16:
This is the repos and tags that are ready to go as 'v7.3.0' of lwa-shell. Or:
```
realtimeradio/caltech-lwa@v2.0.20
realtimeradio/caltech-bifrost-dsp@v1.0.4
lwa-project/ovro_data_recorder@main
ovro-lwa/lwa-pyutils@v1.4.5
ovro-lwa/lwa-antpos@0.6.9
ovro-lwa/mnc_python@0.8.8
ovro-lwa/dsa110-pyutils@v3.8.1
```

jack-h, 2023-06-16:
Was that message to me @rh-codebase ? When I select the GPU nodes for deployment via the web gui I can tell it is running the `deploy` script in the `caltech-bifrost-dsp` repo (the web GUI showed errors because there were some old root-owned build files in the repo which I needed to delete for the deployment to complete).

rh-codebase, 2023-06-16:
thx. I'll keep that in mind when debugging.

rh-codebase, 2023-06-22:
the repo failures I was able to reproduce related to a tagged commit which diverged from the master branch in dsa110-pytuils repo. A more strict workflow is needed. The most robust one is to ensure user branches rebase on top of main(master) so that pulls only involve a fast-forward. The tag can be applied at the tip of the rebase before fast-forward if desired since the fast-forward will not create a new commit hash. All seems to be working as designed when deploying to the MCS host. The next test will be to deploy on the GPU's. None of my testing revealed how the deploy script could be running in a repo not listed in the config file on the GPU host. When I looked, Jacks repo was not in the config. I'll try to reproduce when we deploy to the GPUs. Actually, would be best to isolate 1 GPU for the initial testing.

caseyjlaw, 2023-06-28:
I successfully used the webUI to deploy `v7.3.9` of the shell. This includes adding `ovro_data_recorder` and repos listed above with `deploy` scripts (e.g., `caltech-bifrost-dsp`). Thanks all for helping us get to this point!

@jaycedowell To avoid confusion, we should migrate services to using the code managed under this repo, rather than the one in the home directory. How would you suggest doing that migration?

rh-codebase, 2023-06-28:
@caseyjlaw It still looks like you are not rebasing on top of latest main. I'm seeing merges not fast forwards.

rh-codebase, 2023-06-28:
also I'll be redoing the deploy in a minute so it uses the latest service. I still need to debug this framework.

jaycedowell, 2023-06-29:
> @jaycedowell To avoid confusion, we should migrate services to using the code managed under this repo, rather than the one in the home directory. How would you suggest doing that migration?

@caseyjlaw that should happen automatically when the deploy script calls `generateServices.py` to build the data recorder service files.

caseyjlaw, 2023-08-03:
@rh-codebase Can I help with testing the deployment framework?