ovro-lwa/lwa-issues #300: Migrate docs to ovro-lwa.github.io

caseyjlaw, 2023-03-08:
Documentation should be migrated to public locations, either with static html (`ovro-lwa.github.io` repo) or github wikis linked from that page.

Steps:
* Move subsystem docs to readthedocs
* Create recipes and tutorials in static html
* Create page of observing issues and solutions
* Develop memo series and link from landing page

We'll need to deprioritize SharePoint docs once new ones are in place.

rh-codebase, 2023-03-08:
Let's not forget unit tests. Not only do these serve to test functions but they themselves are excellent tutorials on how to use the function.

caseyjlaw, 2023-03-08:
@rh-codebase I'd like to include an updated diagram of the network. There were some changes over the last year or two that I tried to capture.
Can you confirm that the details are correct at https://docs.google.com/drawings/d/1XXdJOfNHl_aI5B-8zZWnoU4qSgZ2PvUWQtvHsSsO5T8/edit?usp=sharing?