ovro-lwa/lwa-issues #303: ssh among nodes in LWA shelter

ldaddario, 2023-03-14:
[Submitting this at Casey's suggestion.]
When logged into any calim node or digital subsystem node, ssh and scp to other nodes fails.  

Conditions:
- Initial login is from outside ovro via 3-hop public key forwarding:  ssh.ovro.caltech.edu -> lxdlwamaas -> (destination).
- Internal ssh/scp command includes -A switch
Results:
- From any calim node or digital rack node to a digital rack node:  "Permission denied (public key)"
- From any digital rack node to any calim node:  prompts for password.
- From any calim node to another calim node:  authentication succeeds.

Examples:

(casa) ubuntu@lxdlwagpu09:~$ ssh -A lxdlwacr
ubuntu@lxdlwacr: Permission denied (publickey).

(casa) ubuntu@lxdlwagpu09:~$ ssh -A lwacalim00
ubuntu@lwacalim00's password:

(base) [pipeline@lwacalim00 ~]$ ssh -A lxdlwacr
pipeline@lxdlwacr: Permission denied (publickey).

(base) [pipeline@lwacalim00 ~]$ ssh lwacalim09
Activate the web console with: systemctl enable --now cockpit.socket
Register this system with Red Hat Insights: insights-client --register
Create an account or view all your systems at https://red.ht/insights-dashboard
Last login: Tue Mar 14 22:15:41 2023 from 10.40.0.20
(base) [pipeline@lwacalim09 ~]$

rh-codebase, 2023-03-14:
@ldaddario It may help @kembyct if you identified your OS and what ssh client you are using to get into OVRO.. I noticed in the output posted that a password was needed. That indicates that your ssh agent is not forwarding your credentials. The agent is local on the originating machine.

ldaddario, 2023-03-14:
On 3/14/2023 15:45, rh-codebase wrote:
> @ldaddario <https://github.com/ldaddario> It may help @kembyct
> <https://github.com/kembyct> if you identified your OS and what ssh
> client you are using to get into OVRO.. I noticed in the output [...]

Rick,

Please read the posting more carefully.

--Larry

-- 
=======================================================
Larry R. D'Addario
Department of Astronomy
California Institute of Technology
Pasadena, California 91125

291 Cahill, mail code 249-17
email:  ***@***.***
phone:  +1/626/395-1431 (office)
         +1/626/510-6492 (home)
         +1/626/657-6357 (mobile)
=======================================================

rh-codebase, 2023-03-15:
Hmm. I just don't see a mention of the OS of the originating host machine. I tested your examples above using Linux - Ubuntu 16.04 and my ssh client is:
[18:39:52] 255 $ ssh -V
OpenSSH_7.2p2 Ubuntu-4ubuntu2.4, OpenSSL 1.0.2g  1 Mar 2016

and no issues at all. I can ssh between machines and my ssh agent is forwarding my passphrase every time.

yupinghuang, 2023-03-15:
Rick, you're the only one whose key has access to the ubuntu user on calim. Everyone else ought to be getting their own named account on calim

ldaddario, 2023-03-16:
On 3/16/2023 11:27, Casey Law wrote:
> It looks like I can do this ssh with some care on key forwarding. So
> 
>  1. ssh lwamaas
>  2. ssh -YA lxdlwagpu09
>  3. ssh ***@***.***

Casey,

This does not work for me.  I had already been using the -A flag (for 
agent forwarding); the -Y flag is for X11 forwarding, but I'm not using 
X11.  I nevertheless tried -YA and it doesn't help.  From lwamaas to any 
digital subsystem node gives "Permission denied (public key)" and to any 
calim node it prompts for a password, just as I said in the github issue 
posting.

--Larry

-- 
=======================================================
Larry R. D'Addario
Department of Astronomy
California Institute of Technology
Pasadena, California 91125

291 Cahill, mail code 249-17
email:  ***@***.***
phone:  +1/626/395-1431 (office)
         +1/626/510-6492 (home)
         +1/626/657-6357 (mobile)
=======================================================

caseyjlaw, 2023-12-14:
At least two of the ssh commands in the original post are trying to log in as user ubuntu to servers with no such account.
Supported accounts and connections:
- from ovro gateway to ubuntu@lwamaas, then
- <username>@lwacalim[xx]
- some may do ubuntu@lxdlwagpu[xx]
- some may do ubuntu@lxdlwacr (but really only Kathryn).

I'll close, but feel free to reopen is any of these do not work for you.

closed 2023-12-14 by caseyjlaw